pi1541/Pi1541 #3: Add lcd/oled support over i2c and spi

penfold42, 2018-05-18:
And add rotary encoder support for quick navigation

pi1541, 2018-05-30:
I have the SSD1306 displaying data.
Now I just need to integrate it into the browser.

pi1541, 2018-05-30:
penfold42 has generously given me a 128x32 to work with.

pi1541, 2018-05-31:
Scratch that, its a 128x64.

andremiller, 2018-05-31:
Thank you for the feedback. I've ordered a few so I can play along when you're ready :-)

tebl, 2018-07-23:
[image]

Tried the sh1106 driver, but from what I can tell it's an entirely different driver so it just displays random pixels - the display I have soldered up to it is a SSD1306 128x32 (added a picture). From what I can tell the code assumes an atleast 64 high display so probably should have checked that one before soldering a smaller one to the board, though it is still somewhat readable :-P

penfold42, 2018-07-23:
I think we can improve it

Can you build from source ?

Try changing in src/SSD1306.cpp

SendCommand(SSD1306_CMD_SET_COM_PINS);	// 0xDA Layout and direction
	SendCommand(0x12);

To
SendCommand(SSD1306_CMD_SET_COM_PINS);	// 0xDA Layout and direction
	SendCommand(0x02);

You should then have a readable 2 line display

tebl, 2018-07-25:
[image]

Now we're talking, that seems to have solved the problem of the display being squished so driver-wise it seems good. However, it seems to only list the last two of the previously four lines of the display so still a ways to go. Assuming it's just a few assumptions somewhere in the code.

Had one heck of a time getting things built though, seems my bleeding edge 14.04 ubuntu isn't so recent anymore. Managed to build via docker, attaching the files I used in case someone else needs them (unpack in repo, paths assume it's at ~/raspberry/Pi1541).

[Dockerfile.zip](https://github.com/pi1541/Pi1541/files/2228147/Dockerfile.zip)

penfold42, 2018-07-25:
Cool ! I’ll add the options.txt for the 128x32 display.

I can also fix the top2 va bottom 2 easily

I also need to fix the display height so moving up and down don’t scroll past the visible area

penfold42, 2018-07-26:
Can you please try my branch ?

https://github.com/penfold42/Pi1541/tree/128x32?files=1

penfold42, 2018-07-26:
Actually, you can just pull the latest master as it’s been merged in

tebl, 2018-07-26:
[image]
[image]

Listing is working perfectly now, somehow even feels a lot quicker to navigate than previous versions. The only part that seems to need any tweaking is emulation mode with multiple disks, I'm assuming this one is still set to use four lines with the result that the second and possibly subsequent disks remain below the screen.

penfold42, 2018-07-26:
Damn! Missed that one. I’ll look into it.
All the code assumed 128x64 and 4 rows of 16 point font.

It should be faster - I2C is slow and now it only has to do roughly half the work

penfold42, 2018-07-27:
fixed i think...
https://github.com/penfold42/Pi1541/tree/128x32fix

Can you please test and we can get this included in the next build

tebl, 2018-07-28:
Tried out the branch earlier today and it works great! Though it was probably not intended that way, it also looks good on a 128x64 display with a sort of interlaced effect. Nice extra, accidental feature :-)

penfold42, 2018-07-28:
Excellent.

I agree it a cool effect - it’s the reverse of the squishy machine you got using a 128x32 with software for the 128x64

penfold42, 2018-08-02:
Closing as i2c is done and spi is somewhat underway

closed 2018-08-02 by penfold42