pi1541/Pi1541 #8: FB64: Copies files from directory to root

thErZAgH, 2018-05-28:
Hi. 

My structure of the SD card looks like this:

/1541/FB64
/1541/dir1/1.d64
/1541/dir2/2.d64
/1541/dir2/3.d64
...

If I use FB64 it copies n.d64 to /1541/

So I have 
/1541/FB64
/1541/1.d64
/1541/dir1/1.d64
....
if i choose 1.d64.

Why?

pi1541, 2018-05-28:
Sorry I don't know what you mean.

Are you writing back to the disk? And when it saves it saves into the wrong folder?

thErZAgH, 2018-05-28:
Hi.

I'm just choosing an image out of an Directory...

andremiller, 2018-05-28:
I've also noticed something similar, but need to play with it more to determine under which conditions it happens.

In my case I noticed a gamename.d64 in my /1541 directory and I'm fairly certain I did not copy it there. I did play that game earlier in the day and it was mounted with FB64 from /1541/c64/games/tested/gamename.d64. I believe this game does write highscores back to disk.

thErZAgH, 2018-05-28:
Hi.

Ahhh. Could it be that happens wo prg/images that writes some data!? Highscores, user data (quantum reloaded for e. g.), config files...

Hmmmm.... Good point....

thErZAgH, 2018-05-28:
*ro

thErZAgH, 2018-05-28:
*to prg/images

pi1541, 2018-05-30:
Tracked it down to;-
1. If you mount a disk with "OnResetChangeToStartingFolder = 1" in the options.txt.
2. Write/save to the disk.
3. Reset the computer.

It will detect the reset change back to the 1541 folder THEN write the file back to the wrong folder.

pi1541, 2018-05-30:
Fixed

closed 2018-05-30 by pi1541