pi1541/Pi1541 #13: C128 hangs on boot/reset

penfold42, 2018-05-30:
If I unplug the iec cable it starts fine
If I hold C= tonforce 64 mode it’s also fine.

C128 supports autobioting - I wonder if it’s getting confused  by the Pi

pi1541, 2018-05-30:
Last time I checked it was working. I will have to go through old builds and see when it broke.

penfold42, 2018-05-31:
I also need to revalidate that this happens and how/when. Mainly wanted to flag it in case others saw it 

It’s a 128dcr with jiffydos and the internal drive is remapped to drive 11

wernersv, 2018-06-26:
I can see this too when using HW option B. As soon as my C128DCR accesses a drive the system will hang. Cutting the power to the 7406 will continue the disk operation. There is no problem when using HW option A. 

1571 device 8
PI1541 device 9

wernersv, 2018-06-28:
@penfold42, I had similar issues with mine created using prototyping board. Accessing the internal drive in either c128 or c64 mode did not work until the cable was unplugged. This was caused by some wires shorting. The short was fixes and it worked fine.

xlar54, 2018-08-10:
Bump.  It freezes up completely on me in 128 mode (as if trying to autoboot).  Only 64 mode currently works.  Standard flat 128, no JD

penfold42, 2018-08-10:
Hardware variant ?
Mines split lines

pi1541, 2018-08-12:
Should be fixed in V1.12
It was trying to load the boot sector via direct access commands.

penfold42, 2018-08-12:
Happy here !

Thanks

closed 2018-08-12 by penfold42