pi1541/Pi1541 #15: No piture on SSD1306 display

thErZAgH, 2018-06-06:
Hi Peter.

Some ppl -- including me -- are asking how to connect the SSD1306 display on RPi3 the right way.

I connected VCC 3,3V, GND, SD and SC. SD on PIN (not GPIO) 27 and SC on PIN (not GPIO) 28.
My option.txt:

_LCDName = ssd1306_128x64
i2cBusMaster = 0 //SDA - pin 27 SCL - pin 28_

No picture. 

I bought this one:

https://www.amazon.de/AZDelivery-Display-Arduino-Raspberry-gratis/dp/B01L9GC470/ref=sr_1_1?ie=UTF8&qid=1528315648&sr=8-1

penfold42, 2018-06-07:
Weird - I just tried mine with the latest git pull and it worked.

I tested both bus 1 and bus 0.

I2c needs PullUp resistors on the bus to work. Can you check if your display has resistors between sda and VCC and also between scl and VCC ?

penfold42, 2018-06-07:
I’ve done some digging - I suspect many cheap Ebay ssd1306 do NOT include pull up resistors on sda and scl. Mine doesn’t.

I fiddled with mine and there was some corruption when used with pins 27/28.

The Pi HAS pull-ups on the spi bus on pins 3/5 but DOES NOT on 27/28.

I soldered on a pair of 4k7 pull up resistors and now the corruption appears to be fixed on mine.

jgilcas, 2018-06-07:
Could be but at the moment mine is working in 27/28 only changing software software... mmmm i will try to add 4k7 pull up :)

[image]

jgilcas, 2018-06-07:
All used by me had 0x3C, i used them in txzduino and tapuino :)

penfold42, 2018-06-07:
The ssd1306 can only do two different i2c addresses - 0x3c and 0x3d.
Most have a resistor to select between them but it is marked as 0x78 / 0x7a (its the value shift left 1 bit)

As an aside, I’ve just created a PR to allow specifying the address in options.txt

thErZAgH, 2018-06-07:
That could be another reason not to get a picture...

1c3d1v3r, 2018-06-07:
If you have a logic analyzer you can use it to check for an ACK (slave response to I2C communication, correct address etc.)

thErZAgH, 2018-06-07:
Problem could be solved with 1.04 maybe. Will give a try if I'm at home in about one hour.

jgilcas, 2018-06-07:
Versión 1.04 is working perfect for me, brilliant :) 🥇

thErZAgH, 2018-06-09:
I kicked my compiled code I did myself. Now it works.

closed 2018-06-09 by thErZAgH