pi1541/Pi1541 #19: Formatting of D64/G64 not working

thErZAgH, 2018-06-11:
Hi.

Me and another guy testet the basic funtions of a 1541: Formatting disks. D64 and G64. I tried to format an existent D64 and G64. If I show directory afterwards it seems to be working. But after switching C64 and Pi1541 off and on and re-showing directory all files are still there. No formatting has happend!?

Am I missing aomething?

thErZAgH, 2018-06-11:
It seems that the disks are write protected...

penfold42, 2018-06-11:
Did you try unmounting the image before poweroff ?

I’m not sure changes are written real time

thErZAgH, 2018-06-11:
Unmounting? Manually? Uhm... With ESC? 

I just asked about unmounting in Lemon64. I'm unsure what happens after doing some writing stuff to disks...

penfold42, 2018-06-11:
Yes, with escape.

pi1541, 2018-06-11:
When emulation exits and a disk has been modified and the disk image is not read only on the SD card then the modified image will be written back.
When writing back a track cannot be made longer than the original disk image's track. So formatting an existing G64 image may not be guaranteed.

pi1541, 2018-06-11:
If you want to create a new blank d64 just issue the format before entering emulation. The Pi will create a new D64 of the name given in its current folder.

thErZAgH, 2018-06-11:
Thank you. It works (creating a empty D64 in non emulation mode).

closed 2018-06-11 by thErZAgH