pi1541/Pi1541 #22: One line more then expected

thErZAgH, 2018-06-14:
Hi.

I looked into the code. But I don't know where the place for y (start)position is. Sometimes, if the char has 15 (16?) pixel height, a line of the white bar displays at the upper screen.

[https://www.forum64.de/index.php?thread/81832-neue-laufwerksemulation-pi1541/&postID=1272547#post1272547](url)

pi1541, 2018-06-14:
I can't reproduce it on my screen.

penfold42, 2018-06-14:
I did see it previously- possibly before I moved the i2c pins.

What i2c pins are you using ?
If using 27/28 Have you added pull-up resistors ?

thErZAgH, 2018-06-14:
27/27 /wout pull-up. 

Other guys have this phaenomena, too.

penfold42, 2018-06-14:
I2c corruption due to insufficient pull up resistors.

The cheap Ebay displays don’t have them and the pi doesnt on 27/28.
The Pi _does_ have them on the other i2c bus.

thErZAgH, 2018-06-14:
Uhm. Ok, 4,7kO, or does two 10kO the job, too? I have only one 4,7kO.

penfold42, 2018-06-14:
I used 4k7 resistors on each SDA and SCL.

Use your 4k7 for one of them and put 2x10k in parallel for the other

penfold42, 2018-06-14:
I just noticed mine is doing this. I’m using the  i2c on pins 3/5 and I have pull-up resistors.

If I slowly navigate with single keyboard presses it’s fine.
If I HOLD the down key to scroll continuously down, I see this issue

penfold42, 2018-06-15:
And now mine isn’t doing this.

It’s an intermittent issue.

On reset, sometime the display is corrupt. If it is, I always get the extra line.
If it’s not corrupt at start, then it’s fine.

I notice the ssd1306 data sheet has a software init flow diagram for which registers need setting up and in what order - I’ll check this against the code.
Another possibility is the display needs some delays during init.

thErZAgH, 2018-06-23:
Could this be **one of the init problems** of the SSD1306?

This line

`  SendCommand(SSD1306_CMD_ENTIRE_DISPLAY_ON);`

is not a the end of init routine in

`https://github.com/pi1541/Pi1541/blob/master/src/SSD1306.cpp`

Could this be a problem?

penfold42, 2018-06-23:
It is a mystery - I’ve noticed the pi1541 splash screen always looks ok.

My guess is some of the other hardware init code could be causing spurious junk to be sent

penfold42, 2018-07-06:
#38 includes a workaround for this.

I need to find time with the logic analyser but for now I reissue the oled init commands again after all the other init stuff has happened but before the browse list pops up

penfold42, 2018-08-08:
i jus noticed this in IEC::BUS Initialise

// Clear all outputs to 0
		write32(ARM_GPIO_GPCLR0, 0xFFFFFFFF);

I wonder if this is glitching the I2C bus...

pi1541, 2018-08-12:
Perhaps, try calling RPI_I2CInit before any write32(ARM_GPIO_GPCLR0, 0xFFFFFFFF); is called.

penfold42, 2018-08-12:
Tried that, also tried changing mask to fffff0

The problem I’ve now got is trying to reproduce it reliably - even removing my workaround re inits