pi1541/Pi1541 #28: Add sh1106 oled support

penfold42, 2018-06-21:
I will, when it arrives

andremiller, 2018-06-21:
Which one are you planning on adding @penfold42 ? I see there's one with multiple interfaces (3-wire SPI, 4 wire SPI and I2C) and another that only has I2C.

penfold42, 2018-06-21:
I2c

penfold42, 2018-06-23:
Anyone out there got a sh1106 based display ?

pluto63, 2018-06-29:
I do have the i2c 4 pin 1.3” oled, as for now it showed some garbage on startup then black. I’m using this type of oled on tzxduino and code needed to be modify for compatibility.

penfold42, 2018-06-29:
Did you change the lcdname to sh1106_128x64

pluto63, 2018-07-01:
Yes I did change it and it doesn’t work. Just some garbage for a second at startup.

penfold42, 2018-07-01:
ok, thanks - I've ordered one so Ive done no real testing - just took a punt based on the datasheet

penfold42, 2018-07-16:
#41