pi1541/Pi1541 #31: No automatic boot of D64

thErZAgH, 2018-06-24:
Hey Steven.

What does "reset" mean? I must load fb64.d64 manually all the time. Gimme a hint :).
And: The parameters in the option.txt has to be in what order? If I set autoMountImage=fb64.d64 at any place the Pi1541 doesn't boots up. If I place autoMountImage=fb64.d64 at the end of the option.txt it boots. But it does not load fb64.d64 automatically.

I have the latest build 1.07.

pi1541, 2018-06-25:
Order should not matter.
It will auto mount the image. You still need to load it manually.
For example, with AR6. 
At its boot screen press F7 - INSTALL FASTLOAD
Then F1, then RETURN (FB64 will load)
Once FB64 has loaded select <-

When I get time I will do a video and post it on YouTube.

penfold42, 2018-07-09:
Did you get this working ? It works for me.

You may need to wait a few secs after the Pi boots up.

If not, can you paste your config.txt and a dot listing of the 1541 folder

tmr0, 2018-07-17:
It works for me.  I can autoboot CP/M on a Commodore 128 from a d64 image.