pi1541/Pi1541 #32: Confusing similar name files.

pluto63, 2018-06-29:
Having 2 similar names in a directory only defaults load to one, both using filebrowser or direct command load. 
Ex. File : commando.prg and File: commando 2.prg in the same directory always load the commando 2.prg file whatever you choose.

penfold42, 2018-06-30:
Does it also fail if there’s no space on the name ?

pi1541, 2018-06-30:
I cannot reproduce the issue.
FB has a limitation of 16 characters for a file name.

CBM-FileBrowser can only determine if it is a disk image if the extention is in the name.
So for files that are too long we stomp the last 4 characters with the image extention and pattern match it.
This basically changes a file name from something like SOMELONGDISKIMAGENAME.D64 to SOMELONGDISKIMAGENAME*.D64
so the actual SOMELONGDISKIMAGENAMETHATISWAYTOOLONGFORCBMFILEBROWSERTODISPLAY.D64 will be found.

I don't want to use short filenames like COMMAND~1 (like SD2IEC does) as in my opinion this is worse.
Unlike IBM DOS, CBM DOS can handle 16 character filenames. If you keep your filenames under 12 characters everything should be fine.

pluto63, 2018-07-01:
Probably I’ve explain it wrong, to reproduce it use single file .prg as mention in the example above, no need to use filebrowser just be in the correct directory and issue a load command with exact file name, it will load the first one in the list order of the fat 32 directory. The example I made was with way shorter than 12 characters file name. So in my directory “commando 2.prg” is first and “commando.prg” second, in any way FB or direct load will always load commando 2 file. Will do a video later to show that. Notice it since I also have “cauldron 2.prg” and “cauldron.prg” in the same directory.

pluto63, 2018-07-01:
You can download a small video here showing the issue.
https://drive.google.com/open?id=1W4s4urVlgJSpq18TGmRSGdbqELcZAh_z
It’s kernel related since it performs directly even without FB.

pi1541, 2018-07-02:
Fixed with commit d2b87e609a883f9890a9d51671509afe772b62ca

closed 2018-07-02 by pi1541