pi1541/Pi1541 #39: Optimizations to run on pi zero?

mlund, 2018-07-09:
First of all thanks for this grand project! You mention that the code doesn't work on lesser PI's such as the Zero. I *assume* this is due to software emulation being a bottleneck, and I wonder if you would able to provide hints/profiling as to where to set in with potential code optimizations? If appropriate, I'd be happy to help with this part.

pi1541, 2018-07-11:
I have added low level profiling metrics (performance.c and h)
For an example see Drive.cpp (uncomment #define PROFILE 1)
Play around with the different types of performance counters in different parts of the code.

To see the output.
Add -DDEBUG to CFLAGS in Makefile.rules
Hook up a terminal to the PI's serial port (pins 8 and 10) 
115200 Baud
8 Data bits
1 Stop bit
No Parity

As far as I can tell when emulating on a Pi0 it takes 2us most of the time. (see commented code in main.cpp line 984)

I don't really know where the bottle neck is. I assume most of the update loop is in Drive.cpp (as this emulates the 16mhz stuff). Maybe the GPIO queries are too slow on a Pi0.

As we are not using the GPU I don't know if the shared L2 cache size can be changed to give the CPU more. 

I have tried stripping everything back for the Pi0 (ie no buttons, screens etc) and it is still too slow. It will work in SD2IEC mode but as soon as a disk is mounted and the 16mhz stuff kicks in, it cannot do it quick enough. I have tried splitting the updates for the Dive code into separate read and write updates but this made no real difference.

Thanks for the help!

RaymondDay61, 2018-10-12:
I don't know maybe it all ready is in machine language. That would run fast.

pi1541, 2018-10-17:
Hi Raymond,

Thank you very much for your donation, I really appreciate it.

I can't get the @$ command to work in browse mode. It is loading the
directory and attempting to send it but the 64 starts going crazy with the
ATN line.
The /$ command works.

SuperCard Plus was created by Jim Drew and I would like to stay away from
it.
Jim is developing his own 1541 emulator that he is going to sell.
I kind of stole Jim's thunder by making mine open source.

Thanks again,
Steve

fordp2002, 2019-03-06:
Just to throw a few hints from another retro project which I have been watching there are lots that can be done. 
1) You need to make sure that the cache is enabled.
2) You can lock contents in the cache so the bits that need to be quick can be consistently quick
3) It is possible to use the GPU as a parallel processor.

The project in question is this https://github.com/hoglet67/PiTubeDirect and I did NOT discover or implement any of the cleaver stuff above. There is a bit of my code in a couple of areas in that project but nothing related to Pi performance!

andreacampanella, 2019-03-06:
Maybe @hoglet67 can help you debugging the problem.

jigpu, 2019-05-21:
If the 16MHz code in Drive.cpp is the bottleneck (I unfortunately don't have a Zero to test), would it be possible to reduce the work done? Two optimizations come to mind after looking at the code:

1. Change the various floats (`cyclesForBit`, `cyclesPerBit`, and `randomFluxReversalTime`) to integer type. The Pi should have hardware floating-point, but ints would still be faster and the precision shouldn't be necessary. Truncating the first two to integer precision would introduce a slight timing error on the order of 1.5% (rounding would have an error of ~1%), but that should be within the mechanical tolerances of the drive. Changing `randomFluxReversalTime` to something like `randomFluxReversalCycles` should also be fine and save the occasional expensive floating-point division.

2. Jump over cycles that don't need to be emulated. Depending on how fast (or slow) jumps/branches are, this *might* take more time than the current `for (int cycles = 0; cycles < 16; ++cycles)` loop, but something like the following would advance `cycles` and other loop variables directly to the next actionable state:
    ~~~
    int cycles = 0;
    while (cycles < 16) {
        int cyclesToNextInstruction = 16 - cycles;
        int cyclesToNextBit = cyclesPerBit - cyclesForBit;
        int cyclesToNextCarry = 0x10 - UE7Counter;
        int skip = min(cyclesToNextInstruction, cyclesToNextBit, randomFluxReversalCycles, cyclesToNextCarry);
        cycle += skip
        cyclesForBit += skip;
        randomFluxReversalCycles -= skip;
        UE7Counter += skip;
    
        /* Body of the original 16MHz for loop */
    }
    ~~~