pi1541/Pi1541 #43: Feature request: Create blank disk

tebl, 2018-07-18:
Originally wanted to build the Pi1541 so that I could use it in combination with my 1541-drive in order to back up my disks, it would be nice to have the ability to create new blank disks on the fly and add it to the list (just named based on a incrementing counter, that way no need to additionally mess with time servers). Perhaps by holding down the add-switch or just pressing a keyboard button.

penfold42, 2018-07-19:
Sounds like a good idea - I remember doing something similar with a zoomfloppy and shell script.

I've started looking into it - ive already refactored some code to reuse the "create.d64" function

penfold42, 2018-07-20:
Done

penfold42, 2018-07-27:
Have you tried this yet ?

closed 2018-10-17 by pi1541

hectorezim, 2020-12-19:
I am interested on using this feature, what do I have to do to create a new disk on the fly on it?