pi1541/Pi1541 #46: Rotary encoder support

penfold42, 2018-07-21:
Instead of up and down separate buttons.

Center button should enter folder, run d64 or lst file

gbouille, 2018-12-31:
Hi,
I succeded code a rotary button support.

But:
The rotary button I use is kind of button with a clk and dt signals.
The sequence for a clockwise step is : clk 0->1 then dt 0->1 then clk 1->0 then dt 1->0.
The sequence for a reverse clockwise step is : dt 0->1 then clk 0->1 then dt 1->0 then clk 1->0.
The speed of the sequence depend of the speed you turn the button.

So to use it, you have to detect the upfront on clk signal and the dt signal indicate the rotation (1 for clockwise and 0 for reverse clockwise by example).

The main issue is that the way pi1541 works is to pull the button i/o regulary. So if the rotation step sequence is too fast, you can detect the clk upfront too late and faultly detect the dt line. You can even completly miss the clk signal.
So it work only when you slowly rotate the button.

The only way to have a more robust rotary button support without changing the way pi1541 works is to use another type of rotary button using quadrature encoding where a step sequence only change one of the entry once.
The other solution is to use interrupt to detect the clk edge. But you have to change the pi1541 deeply.

gbouille, 2019-01-03:
Hi,
I've implemented a interrupt based rotary button support.
Check my GitHub branch here : https://github.com/gbouille/Pi1541/tree/rotaryEncoderSupport
It's works much better than the previous implementation.
Let me know if you that I make a pull request.