pi1541/Pi1541 #49: Feature Request: ALT+A to reset drive to AutoMountImage?

MeAndMyRhythmbox, 2018-07-26:
I share one display with C64 and PI. Using a disk image with FB64 as recommended let's me browse without PI screen and use a fast loader cart... However I have to power cycle the C64 to reset the emulated drive, which then loads the automount image (simply resetting the C64 leaves the emulated drive unaffected, which is probably the preferred behaviour.) Am I missing a trick here? 

i.e Is there any way to reset the emulated drive without power cycling the computer?

If not, can we have a blind shortcut (e.g ALT+A) to reset the emulated drive and save some power power switches? Thanks in advance :)

penfold42, 2018-07-26:
We could also make escape do this optionally via a options.txt directive. 

This was its behaviour until it annoyed me !

Maybe we do both.

MeAndMyRhythmbox, 2018-07-27:
Haha, that explains it... I thought something had changed.

Don't think we want the escape key configurable, since current behaviour (dropping out of emulation back into browse mode) makes sense.  A dedicated key combo to reset the emulated drive (ALT+R) or automount default drive image (ALT+A) would be clear and really helpful. A combo using the GPIO push buttons would be a bonus so USB keyboard not a necessity. Maybe button 1 and 2 together?

Happy and grateful for whatever is easiest / quickest to implement :)

penfold42, 2018-07-27:
Alt-A would be good in both browse and emulation mode.

Should Alt-R basically fake an IEC bus reset ?

MeAndMyRhythmbox, 2018-07-27:
Yeah, assuming IEC bus reset is what happens when C64 is power cycled...

Thus ALT+R (and maybe GPIO buttons 1+2) could initiate a drive reset to give user default folder to browse or default disk image to boot with fastload cart (dependent on options settings.) Useful to be back in default state without having to power computer off/on.

penfold42, 2018-07-27:
I had a spare 10 mins so I’ve done Alt-A in browse mode. It works well for quickly reloading the auto image 

Will tackle emulator mode Alt-R tomorrow.

MeAndMyRhythmbox, 2018-07-27:
Star! Many Thanks.

penfold42, 2018-07-28:
I havent done Alt-R because I'm not sure what would be different from hitting escape?
I have however extended Alt-A to emulation mode which is great to quickly re launch the automount image.
#53

MeAndMyRhythmbox, 2018-07-29:
Brilliant! 

ALT-A suits my needs perfectly and enhanced configuration options should suit any other preference people might have. I am on a family holiday all next week, but look forward to testing ASAP :)  

P.S Doesn't escape just exit emulation mode and drop back to browsing the parent folder? Or does it have a function whilst browsing also? Either way ALT-A will help me a lot whilst testing what does and does not work with my machines. Lounge TV takes for ever to switch between C64 and PI inputs!

MeAndMyRhythmbox, 2018-07-29:
Had a chance to try build v1.10 just now and ALT-A is working perfectly as described :) 

Lots of other goodies in this build also. Great to see the level of enthusiasm for making PI1541 as good as it can be. Still have problems running most demos, but I guess that's vintage hardware for you... Neon by Triad was the reason I bought a PI hat in the first place, but stubbornly refuses to work for me.

penfold42, 2018-07-30:
Excellent - can you pls close this issue ?

closed 2018-08-02 by MeAndMyRhythmbox