pi1541/Pi1541 #67: 1571 / 81 drive images

xlar54, 2018-08-09:
Realizing that the primary goal for the project is a cycle exact 41 emulation, could we implement the ability to r/w 71 and 81 images as well (initially without burst mode of course)? This enhancement has been asked about in various forums, but I wanted to get it on an official issues list.  I may fork it and see if I can help also.

xlar54, 2018-08-11:
not a bad idea.  I did find this code:  https://paradroid.automac.se/diskimage/

penfold42, 2018-08-11:
That could work too. 
 I’d rather incorporate sd2iec cos it brings other goodness with it

xlar54, 2018-08-19:
For now, Im going to have to step back a little.  IEC is a bit above my head at the moment.  But I will happily continue to support this effort.  71 and 81 support are badly needed, and you would beat UII+ to the punch if you could do it :)

pi1541, 2018-10-17:
Most of 1581 emulation is now supported.
Still working on burst mode.

mjg14, 2020-05-30:
Is there any update on burst mode? Or has that been scrapped?

pi1541, 2020-05-31:
Still working on it.
Now days I don't have as much free time to work on the project so that is why things have slowed down, sorry.

mjg14, 2020-05-31:
There's nothing to be sorry for :) . Thanks for your quick reply, and for all the effort you've put into this!

iampowerslave, 2021-02-13:
Hi, any comments on why 1571 is not emulated, while 1581 is?  I do not know the complexity of each of them but supposed that there are more d71 images out there.

penfold42, 2021-02-13:
The 1581 is a bit simpler than the 1541.

the 1571 in someways is an ugly step child of both the 1541 and 1581 so is the most complex of the family

DeeKay64, 2021-06-22:
Yeah, 1541 Ultimate is about to integrate 1581 and 1571 support, too, and while 1581 support was done in a day or so and works like a charm (including burst/fast serial/2MHz!), 1571 support is still quite beta and leaves a *lot* to be desired...

pi1541, 2021-06-23:
It wasn't done in a day. They have been referring to my code and asking me questions for months. Good luck to them.
I have had 1581 and 1571 done for over a year. I did 1581 in a weekend. I haven't fully released the updated code yet as I am still trying to fix a burst bug that sometimes occurs on the turn around from output to input. Sometimes the timing will screw up and it will then stop bursting mid way through receiving data from the 128, gets confused whether it should be bursting or not and then breaks completely. Only happens about 95% of the time.
I have literally spent hundreds and hundreds of hours trying to debug and fix it. Not that people appreciate it. They just go on YouTube and bitch about their free software not working.

DeeKay64, 2021-06-23:
Well, that's what Bart told me at least... Burst mode/fast serial does work with their implementation, and they are making advances on the 1571 front... Would be great to have that on Pi1541, too, cause I love the standalone nature of it (and its display!) and have designed an awesome super compact case for it, too! ;-)

GideonZ, 2021-06-23:
Hi there @pi1541! Are there areas in which I could offer you some help, Stephen?

pi1541, 2021-06-24:
Hey Gideon! Yes I have many questions, thanks.
I will reach out privately. Perhaps through Scott?

DeeKay64, 2021-07-26:
Any news on the C128 compatibility front? ;-) I'd love to be able to use my Pi1541 with my c128... Also, d64 support on c64 does not seem to be fully stable either, can anyone else confirm that Booze's "The Elder Scrollers" crashes at the end of the large dragon scrolling pic on Pi1541? https://csdb.dk/release/?id=179123