pi1541/Pi1541 #75: Configurable PCB buttons S1-S5

Julitium, 2018-08-18:
Hi,
Congratulations for your 1541Pi, a fabulous device.
I need the exchange of functions in the buttons.
in the options file us can configure the function of each one.
example:
s1 = s2
s2 = s1
s3 = s4
....
A simple but very useful function for those of us who have a PCB.

thank you very much.

penfold42, 2018-08-18:
It should be possible. 
I’m curious as to why this helps ?
If you’re designing a pcb, the autorouter can run the traces for you.
If you’re home brewing you can run the 5 wires anywhere.

If we add it, I’d probably use button function names in the options file:
 INPUT_BUTTON_ENTER = 0
 INPUT_BUTTON_UP = 1
 INPUT_BUTTON_DOWN = 2
 INPUT_BUTTON_BACK = 3
 INPUT_BUTTON_INSERT = 4

And maybe use gpio numbers rather than 0..4

penfold42, 2018-08-19:
Actually I’ve found a use for it myself !

I’d rather use my own Pi upside down - I can already flip the display and your suggestion let’s me flip the buttons around

penfold42, 2018-08-19:
done in #77

Julitium, 2018-08-19:
Hi,
I'm very happy about your interest. Thanks.

not just invert the functions ...
also exchange of position on the buttons (any possibility).

there are people who prefer a configuration, and others in another way (our case)

they will also be friendly left-handed people (for example).

about the nomenclature in the options file, I think it is indifferent to adapt it to your liking.

thank you very much.

pi1541, 2018-10-17:
This was made possible in v1.14

closed 2018-10-17 by pi1541