pi1541/Pi1541 #79: No rom found

matozoid, 2018-08-21:
Hi! Sorry for bothering you.

I've followed all instructions, including putting a dos1541 file in the root of the SD card. When booting it says "No ROM file found". I tried to follow the instructions on Linux, then on Windows, then saw that someone introduced a "roms" dir and also put dos1541 in there, but still no luck.

Any pointers?

penfold42, 2018-08-21:
What’s the file size ?

matozoid, 2018-08-22:
```
danny@denkpad:~$ ls -laR /media/danny/PI1541/
/media/danny/PI1541/:
total 3338
drwxr-xr-x  5 danny danny   16384 jan  1  1970 .
drwxr-x---+ 3 root  root     4096 aug 22 18:13 ..
drwxr-xr-x  2 danny danny    1024 aug 21 22:43 1541
-rw-r--r--  1 danny danny   52116 aug 16 11:50 bootcode.bin
-rw-r--r--  1 danny danny      41 sep  5  2017 config.txt
-rw-r--r--  1 danny danny   16384 mrt 31  2017 dos1541
-rw-r--r--  1 danny danny    6659 aug 16 11:50 fixup.dat
-rw-r--r--  1 danny danny  466048 aug 12 17:26 kernel.img
-rw-r--r--  1 danny danny    3847 aug 12 16:28 options.txt
drwxr-xr-x  2 danny danny    1024 aug 21 23:01 roms
-rw-r--r--  1 danny danny 2846532 aug 16 11:50 start.elf
drwxr-xr-x  2 danny danny    1024 aug 21 22:38 System Volume Information

/media/danny/PI1541/1541:
total 38
drwxr-xr-x 2 danny danny  1024 aug 21 22:43 .
drwxr-xr-x 5 danny danny 16384 jan  1  1970 ..
-rw-r--r-- 1 danny danny  2271 jan  6  2018 fb128
-rw-r--r-- 1 danny danny  2359 jan  6  2018 fb16
-rw-r--r-- 1 danny danny  1459 jan  4  2018 fb20
-rw-r--r-- 1 danny danny  2208 jan  4  2018 fb20-3k
-rw-r--r-- 1 danny danny  2248 jan  4  2018 fb20-8k
-rw-r--r-- 1 danny danny  3113 jan  4  2018 fb20-mc
-rw-r--r-- 1 danny danny  2227 jan  6  2018 fb64

/media/danny/PI1541/roms:
total 33
drwxr-xr-x 2 danny danny  1024 aug 21 23:01 .
drwxr-xr-x 5 danny danny 16384 jan  1  1970 ..
-rw-r--r-- 1 danny danny 16384 mrt 31  2017 dos1541

/media/danny/PI1541/System Volume Information:
total 19
drwxr-xr-x 2 danny danny  1024 aug 21 22:38 .
drwxr-xr-x 5 danny danny 16384 jan  1  1970 ..
-rw-r--r-- 1 danny danny    76 aug 21 22:38 IndexerVolumeGuid
-rw-r--r-- 1 danny danny    12 aug 21 22:38 WPSettings.dat
```

```
(parted) print                                                            
Model: SD APPSD (sd/mmc)
Disk /dev/mmcblk0: 62,9MB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 

Number  Start   End     Size    Type     File system  Flags
 1      1049kB  62,9MB  61,9MB  primary  fat16        lba
```

It seems Windows formatted it to FAT16. Maybe because the card is only 64MB?

penfold42, 2018-08-25:
Weird ...

If you can dd /dev/mmcblk0 and gzip it I can take a closer look

penfold42, 2018-08-25:
I just checked mine - it’s fat32

matozoid, 2018-08-26:
Here's a link to the image: https://wetransfer.com/downloads/4e84963be74f11377778415189e5998d20180826180507/30b1980ec56d8d9affa4a3979028f27420180826180507/4ad972

The first card I made was FAT32 by the way. Linux let me choose. Windows did not and just picked FAT16.

penfold42, 2018-08-29:
I just used linux to dd your image to an sdcard and it worked first go.

My card was slightly smaller so of course it complained while writing but it did work.

What brand is your card ? The Pi firmware is clearly happy with it, but it seems the pi1541 code isn’t.

I might look at dumping some diagnostic info to the screen instead of the vaguely epileptic red flashing hang

matozoid, 2018-08-29:
Right, I'll go try some alternatives then. At least I know that the image is probably correct, so thanks for that!

Why not have a guru meditation? ;-)

closed 2018-08-29 by matozoid

onre, 2018-09-19:
I seem to have another card which behaves similarly.

```
.:
total 3324
drwxrwxrwx 1 root root     512 Sep 18 23:49 1541
drwxrwxrwx 1 root root     512 Sep 19 12:34 System Volume Information
-rwxrwxrwx 1 root root   52116 Sep 18 20:30 bootcode.bin
-rwxrwxrwx 1 root root      41 Sep  5  2017 config.txt
-rwxrwxrwx 1 root root   16384 Sep 18 23:35 dos1541
-rwxrwxrwx 1 root root    6660 Sep 18 20:30 fixup.dat
-rwxrwxrwx 1 root root  466048 Aug 12 15:26 kernel.img
-rwxrwxrwx 1 root root    3847 Aug 12 14:28 options.txt
drwxrwxrwx 1 root root     512 Sep 19 12:35 roms
-rwxrwxrwx 1 root root 2846852 Sep 18 20:30 start.elf

./1541:
total 28
-rwxrwxrwx 1 root root 2271 Jan  6  2018 fb128
-rwxrwxrwx 1 root root 2359 Jan  6  2018 fb16
-rwxrwxrwx 1 root root 1459 Jan  4  2018 fb20
-rwxrwxrwx 1 root root 2208 Jan  4  2018 fb20-3k
-rwxrwxrwx 1 root root 2248 Jan  4  2018 fb20-8k
-rwxrwxrwx 1 root root 3113 Jan  4  2018 fb20-mc
-rwxrwxrwx 1 root root 2227 Jan  6  2018 fb64

./System Volume Information:
total 8
-rwxrwxrwx 1 root root 76 Sep 19 12:35 IndexerVolumeGuid
-rwxrwxrwx 1 root root 12 Sep 19 12:34 WPSettings.dat

./roms:
total 16
-rwxrwxrwx 1 root root 16384 Sep 18 23:35 dos1541
```
This is an unbranded 8 GB card. The only identifying marking on it is the text A08Y1712TDA.

matozoid, 2018-09-19:
Similarly means "it works" or "it doesn't work?"

I've ordered a few new SD cards and am waiting for them to arrive.

onre, 2018-09-19:
Similarly aka "does not work". Also just a single partition on the card, tried different types and everything, but didn't get it to work. Just bought a 32 GB Kingston which seems to work just fine.

matozoid, 2018-09-19:
Card arrived, it is now working.