pi1541/Pi1541 #87: Add Tags To Track Releases in GitHub

iNamik, 2018-09-08:
I was interested in learning what code changes happened between v1.11 and v.12.

Coming to the repo and looking for tags showed there weren't any tags configured:

https://github.com/pi1541/Pi1541/tags (empty as of this writing)

Please consider adding tags to track releases to make it easier to understand what code went into each release.

This could also help in reproducing kernel images from source and would make it easier for someone to work on a PR against a known version.

NOTE: You can back-populate tags to commit hashes, but I don't know if you've logged which commit hashes were used to build each official version?

Thank you for your consideration.   Please let me know if you have any questions.

-David Farrell

pi1541, 2018-10-17:
Started doing it for v1.15 onwards.
For older versions just look at the commit log Everytime I changed the build version I did a build then.

iNamik, 2019-05-05:
I'm going to close this issue now as releases have been tagged for months so I would say its done :)

Thanks!

-David

PS: Your most recent release, `1.20` is accidentally tagged as `1.120` -- I'll check back later today and create a separate issue if it still persists.

closed 2019-05-05 by iNamik