pi1541/Pi1541 #88: Clarify Comments for ROM Selection in Options File

iNamik, 2018-09-08:
### Clarify Default Behavior

So the `options.txt` file currently reads:
```
// You can specify additional ROMS here (up to 10)
// You can then use the number keys on the keyboard to change which on you would like to use (only in browse more and not in emulation mode)
// .LST files can also specify one of these ROMS that will automatically selected when the LST file is loaded.
//ROM2 = Jiffy.bin
//ROM3 = d1541II
```

This is unclear which rom would be selected by default in case where:

* I have multiple roms present on the SD card, but have not configured `ROM2` or `ROM3`
* I have Configured `ROM2` and `ROM3` but have 3+ roms on the SD card

The website has the following message:

> ...The ROM must be called dos1541 or d1541.rom or d1541II or Jiffy.bin.

But again it is not clear what order these files are searched in?   I could probably go look at the source code (and maybe I will, to generate PR for this issue) but I think that the comments in the options file should make it more clear.

### Support for `ROM1` Property

Here is how I have configured my options.txt file:
```
ROM1 = Jiffy.bin
ROM2 = d1541II
ROM3 = dos1541
```

The document doesn't mention support for the `ROM1` property, but I added it just to make my intent clear.

### Update Comment for 7-Rom Limit + Function Keys

Comment from v1.11 notes:
> The number keys no longer swap the ROM.
> The function keys retain this feature so use them instead. 
> MUST use F1..F11 for rom and device ID changes. 

Comment from v.1.09 notes:
> Keyboard buttons F8-F11 select device ID 8-11. (Note: amount of ROMS that can be swapped has been reduced to 7, ie keys 1-7)

The comments within the options file should updated to reflect these new rules.

So I think what I am requesting for in this issue is:

* Update the comments in the properties file to clarify default behavior
* Add support for `ROM1` to help the user fully express their preferred order
* Update comments to reflect rom limit (7) and selection keys (F1 - F7)

Thank you for your consideration.  Please let me know if you have any questions.

-David Farrell

pi1541, 2018-10-17:
Fixed in v1.15

kakemoms, 2018-10-25:
Just downloaded raspberry PI firmware and installed pi1541 and rom file on the sd card. Nothing worked. Connected it to my television and it hows "please add dos1541 to root" (or something like it.

I have dos1541 on root, downloaded as part of VICE 3.1. I have tried to rename it dos1541.bin, dos1541.rom and added ROM1=dos1541.bin with no change. It always boot to the same red screen which says its missing the ROM file. So something is broken.

pi1541, 2018-10-25:
@kakemoms 
What you are doing is correct. The file system must be failing to open the file for reading.
Can you please try a different SD card?

kakemoms, 2018-10-26:
@pi1541 
Ok. I tried a different SD card with same results. What I do is copy "start.elf", "bootcode.bin", "fixup.dat" from the file "firmware-1.20180919" into a new sdcard. Then copy "dos1541" from "vice3.1.tar.gz", then copy all files from "Pi1541.zip" (latest version, although there is no version indicator in the content of it). I then copy a couple of .prg (pacman and so on..) into the 1541 directory on the SD card.

I have tested using both FAT and FAT32. Both boot the PI, but end with a red screen saying "No ROM file found!" "Please copy a valide 1541 ROM file in the root folder of the SD card" "The file needs to be called ´dos1541´."

I have also tried to copy ALL drive rom files from vice3.1 to both root and 1541 folder of the SDCard.
I have tried to change options.txt to "ROM1=d1541II without any change (is it even reading options.txt?)
I have tried using diskpart to format the SDCard with id=0c (which is what linux uses for FAT32) without any change to the result.

I recon since both SD cards give the same result and boots the PI, they seems to be ok... but it looks like the PI firmware may have a problem in reading the card. Both cards are of the same type (1GB SDHC cards), so maybe I should get and try a completely different brand. Still, strange that it boots the firmware when it can't read anything else...

I'll get back here with an update if I get it working.

kakemoms, 2018-10-27:
Well, I got it working to a point were I get "searching" on the Vic-20. At that point the PI freezes and nothing more happends. This was achieved by using a different SD Card (Samsung EVO 32GB) at a considerable cost. I guess my OEM cards were not keeping up with the speed at which PI is reading. So now the ROM file is working, although I have other problems (which I will address elsewere).

iNamik, 2018-11-06:
@kakemoms,

 Looks like you may have hijacked this issue for something non-related to the original topic -- Please create a new topic so that both of our issues can get the proper attention.

Thank you.