pi1541/Pi1541 #90: Exit Folder Buttons/Keys Not Working on Kernel v1.12

iNamik, 2018-09-08:
The Exit Folder functions appear to be completely broken on Kernel V1.12

### Devices / Kernel

#### Hardware Tested:

- RPi 3B
- RPi 3B+

#### PI-1541 HATS Tested:

- CommodoreForever v1 (Rev A) (Type B)
- RiBit v1.1 (Type B)

#### Keyboards Tested

- 2 x Wireless
- 1 x Wired

#### Kernel Versions Tested

- v1.11 (Working)
- v1.12 (Not Working)

### Issue : V1.12 : Browser Mode - Exit Folder Not Working

#### GPIO Buttons

The following GPIO Buttons DO NOT work for me in Browser Mode on V1.12:

- Pressing SELECT Button on `..` Folder
- Pressing EXIT FOLDER Button

#### Keyboard Keys

The following Keyboard Keys DO NOT work for me in Browser Mode on V1.12:

- Pressing ENTER Key on `..` Folder
- Pressing BACK Key

### V1.12 : Emulation Mode - Exit Works Fine

Just for completeness, I want to repot that exiting Emulation Mode works as expected.

- Press SELECT Button to exit emulation mode
- Press ESC Key to exit emulation mode

### V1.11 : Browser Mode - Exit Folder Working As Expected

I tested v1.11 and all Exit Folder Buttons/Keys work as expected.

Thank you for your consideration.  Please let me know if you have any questions.

-David Farrell

pi1541, 2018-10-14:
The buttons were fixed in v1.14

iNamik, 2021-01-08:
Closing per statement that the issues was fixed - Thanks!

closed 2021-01-08 by iNamik