pi1541/Pi1541 #92: Clarify USB Port For Keyboard in Documentation

iNamik, 2018-09-17:
I actually thought the keyboard support was broken.  I tried 2 of the USB ports over the course of the first few days of playing with my PI541.  Eventually I decided to just try all 4 ports to see what would happen and one worked!

If the system requires keyboards to be in a specific USB port, let's add that to the documentation somewhere.

For those reading, the (only) port that works for me is:

Lower Left Port  (when facing the USB Ports with Ethernet Jack on the Left Side of the PI)

penfold42, 2018-09-17:
They should all work.

What doesn’t work is hot plugging USB devices - they must be plugged in at power on or reboot

iNamik, 2018-09-18:
Thanks for the note -- I'll confirm that works on my Pi and update/close the issue ...

zomgugoff, 2018-10-12:
I am also having no luck getting keyboard input to work. I have tried all 4 ports of a RPI3 with a wireless Logitech dongle, a wired Logitech keyboard and a wired Dell keyboard. I can see the data indicator update rate at the bottom of the screen drops when I unplug any of them so it must be detecting something.

lagomorph, 2018-12-22:
Is this still an issue? When I plug in an HDMI monitor to see the serial port signals, I can't get a USB keyboard to work. I am using a hat that has an LCD screen and it's also configured to use splitIECLines. The only reason I plugged in a keyboard is because the switches seem to be disabled when the HDMI cable is used. Is that by design? I'd just as soon use the switches and no keyboard as I do without the HDMI connected but neither appear to work.

huskeyw, 2019-04-16:
same issue, keyboard does not work

n9yty, 2019-06-10:
Not working here either, using that lower-left port, or any port, after a power cycle. Just can't get it recognized.