pi1541/Pi1541 #96: sd2iec commands cbmfilebrowser Jaffydos

isidis, 2018-10-15:
Within Jaffydos kernel rom (patched version of Jiffydos including cbm filebrowser). When browsing the content of the SD card every deeper level I browse I get "device not available, error"

perinoid, 2018-10-15:
I can confirm. Browsing works, but that error appears.

SvOlli, 2018-11-17:
This might be a problem that needs a fix in JaffyDOS instead of PI1541. The PI1541 runs a reset when changing a directory, especially when using a disk-image. This blocks the bus, which then JaffyDOS tries to access. CBM-FileBrowser has the same problem, that's why the repo includes a patched version. Maybe it's possible to port the FB workaround to JaffyDOS?

sailor3ad, 2019-04-19:
Isn't this a PI issue since (several) browsers needs fix for them to run on the PI ?

SvOlli, 2019-04-19:
On Fri, 19 Apr 2019, sailor3ad wrote:

> Isn't this a PI issue since (several) browsers needs fix for them to run on
> the PI ?

It's a fault on both sides. Any browser should expect that changing to a 
different directory might not be executed "in an instance" as expected. And 
also some fixing on the PI could be done by something like instead of 
runnning a normal reset, a "frozen" state could be used as a starting point, 
so a reset will not take as long as expected. But the PI option might not 
always work 100% in every cornercase. And implementing that is also not 
trivial when it comes to non starndard firmware like SpeedDOS.

Still I think any browser is making an assumption that "works with sd2iec", 
but that assumption is wrong when using a PI. Since both are not always 
complying to Commodore standards non of them can be considered "the real 
deal".

Winke: o/~
SvOlli
-- 
|  _______       |
| (  /\          | Wesley's dead. I'm feeling grief for him.
|__)v\/lli a.k.a.| I can't seem to control it. I wish to do more violence.
|Sven Oliver Moll|   -- Illyria, Angel Episode 5x22 "Not fade away"

sailor3ad, 2019-04-23:
Solved with http://blog.worldofjani.com/?p=3544