pi1541/Pi1541 #100: Operate Pi1541 alongside real 1541 disk drive

vikingmedia, 2018-10-30:
Hi,
first of all, let me thank you for that great device. It makes using my breadbox so much easier! I wonder if it's possible to operate the Pi1541 alongside a real 1541 disk drive in order to copy files from the Pi1541 to a real floppy. I've bought a kit from ebay and it works perfectly. It even comes with another DIN connector, so I can plug in one of my real 1541 drives. But whenever the real drive is switched on, the Pi1541 works very inconsistently. Sometimes I can load the disk's directory, but most of the time I see a "file not found" error, no matter what I try. Once I turn off the real drive, everything works normally again. Do you have any idea what might be wrong? I'm using release 1.15 and by power supply is 3A, so this shouldn't be an issue.
Kind regards

andremiller, 2018-10-30:
The author has provided two designs to build it yourself ( https://cbm-pi1541.firebaseapp.com/ ), with the second option providing support for multiple devices if you use splitIECLines = 1 in the configuration file.

I think you might have to ask for support directly from the eBay seller you purchased this kit from to see how they implemented it.

penfold42, 2018-10-30:
Did you changed the device ID of the second drive ?

Both real drives and pi1541 default to ID 8 but every drive needs a unique ID

vikingmedia, 2018-11-05:
hi, thanks for your suggestions. The pi1541 is set to device #9, the real 1541 to drive #8. I can access both drives separately using the appropriate device number without a problem. The design of the board is an option 2 implementation, I've set splitIECLines = 1. Both drives kind of work at the same time, but very, very unreliably, as I've mentioned in my original question. The pi1541 sometimes is able to load the directory listing (load"$",9) when the real 1541 is operated at the same time. Most of the times, I get a "file not found?" error.

shazz, 2018-12-01:
I have the same configuration C64 <-> Pi1541 [9] <-> 1541 [8] <-> 1541-II [10] <-> 1571 [11] and I don't have this problem, I can even copy D64 images to real floppies (using DraCopy) without issues. 
Maybe there is a problem with your daughter board ? 

Can you paste your options file ?

mseelye, 2019-01-24:
For clarity, which kit did you buy on eBay? 

I just got a Pi1541 hat on ebay, and if it is the same one I can test to see if I see the same issue.

vikingmedia, 2019-03-10:
Thanks for your replies and sorry for not answering. @shazz, thanks for confirming that it should work. @mseelye , I bought the one from https://www.ebay.at/usr/bruktmoped

Mycowski, 2020-01-21:
Hi, I've exactlly the same problem. @vikingmedia did you already solve it?

vikingmedia, 2020-02-23:
Hi @Mycowski, it works now - all I did was to not use the second DIN connector on the PI1541. Instead, I plugged in the PI1541 on the "Interface" connector on the real 1541, while connecting the real 1541 to the C64 using the "Serial" connector. In other words, the real 1541 is now the master and the PI1541 connected to the real 1541. No idea, why this configuration works, though,

Deathlok2000, 2021-01-20:
The 7406 hex inverter buffer in the PI1541 could be bad. If it is socketed, I'd suggest replacing it with a good name brand chip.

DeeKay64, 2021-06-28:
> 
> 
> The 7406 hex inverter buffer in the PI1541 could be bad. If it is socketed, I'd suggest replacing it with a good name brand chip.

Are you the very Deathlok that made the C*base mod? ;-)

DeeKay64, 2022-10-27:
> I could not get my pi1541 to work with my real 1541 using kernel v1.24. Using v1.23 did work.

yeah, I also went back to v1.23 on my Handheld 64 as recommended by Matthias, who designed the thing.. Mainly because the latest Lethargy demo would not work - it does now on v1.23!