pi1541/Pi1541 #107: Documentation for OLED integration is missing

gbelmont22, 2018-11-20:
I wasn't sure if I should add this as an issue, because it's not a bug - but I hope by submitting this as an issue, the question and answer might help someone else with the same questions I have.  Also, I had no idea where else to ask this...

After successfully prototyping using the (simple) schematics found on https://cbm-pi1541.firebaseapp.com/ I am now in the process of "building" my own pi1541.  I would like to add an OLED display module to my final build, but I haven't been able to locate any documentation on how this is done.

I know it is possible because I noticed several of the Pi1541 knock-offs being offered on ebay are shown with an OLED display.  Further, after snooping around the _Options.txt_ it is plainly obvious there is support:
```
// If you are using a LCD screen then specify it here
//LCDName = ssd1306_128x64
//LCDName = ssd1306_128x32
//LCDName = sh1106_128x64
```

After some googling, I was able to find examples for each of these OLED displays, so I feel confident that if I purchase one of them it will probably work - except I'm not seeing anything that will help me to understand how to physically connect it.

Again, it seems reasonable to assume the LCD connects to the RPi using the i2c bus based on this:
```
// If you are using I2C LCD you can optionally change what pins it is connected to.
// (defaults to 0 for non-split lines (Option A) or 1 for split lines (Option B))
//i2cBusMaster = 0 //SDA - pin 27 SCL - pin 28
//i2cBusMaster = 1 //SDA - pin 3 SCL - pin 5
```

So my question is:

Can you take any of the above (3) OLED displays and connect them to the RPi directly using Ground/Vcc/SDA/SCL without any additional circuitry or level shifting?

and:

Would other displays be compatible, or am I restricted to the three listed? (I'm interested in exploring the trade-offs between size, resolution and the face-plate for my enclosure).

**Finally, I want to thank everyone involved in creating and improving this project.  The pi1541 is a very cool addition to my retro-bench.  Fantastic job guys!**

teknoplay, 2018-11-20:
All the answers to your questions are in the options.txt file, here´s the portion of the file that indicates how to connect the OLED displays.   (My comments between asterisks)
*********** Remove the // comment on the line that points out to the OLED of your choice. *****
// If you are using a LCD screen then specify it here
//LCDName = ssd1306_128x64
//LCDName = ssd1306_128x32
//LCDName = sh1106_128x64

********* This option is useful  to scroll thru the LCD screen ******
// If you are using a LCD screen and you would like PageUp and PageDown keys to work with it then specify this option
//KeyboardBrowseLCDScreen = 1

******** This option is cool!  *************
// change startup logo on oled - 1541ii or 1541classic
//LcdLogoName = 1541ii
//LcdLogoName = 1541classic
//LcdLogoName = customfile.raw

******** Here´s the important section, depending of which option you build (A or B) the  OLED 
is connected to different Pins on the GPIO. You don´t need anything else than connect the correct
pins, the OLED comes with pull up resistors. I connected both models and they work very well.*****

// If you are using I2C LCD you can optionally change what pins it is connected to.
// (defaults to 0 for non-split lines (Option A) or 1 for split lines (Option B))
//i2cBusMaster = 0 //SDA - pin 27 SCL - pin 28
//i2cBusMaster = 1 //SDA - pin 3 SCL - pin 5
//i2cLcdAddress = 60	// I2C display address in decimal and shifted. 60 == 0x78, 61 == 0x7A
//i2cLcdFlip = 1 // Rotate i2c LCD screen 180 degrees
//i2cLcdOnContrast = 127 // Allows you to adjust the contrast on your i2c LCD screen
//i2cScan = 1 // scan i2c bus and display addresses on screen

pi1541, 2018-11-25:
> Can you take any of the above (3) OLED displays and connect them to the RPi directly using Ground/Vcc/SDA/SCL without any additional circuitry or level shifting?

Yes

penfold42, 2018-11-25:
As long as you power the display from 3.3v and NOT 5v

Pentasis, 2022-09-09:
Sorry to ask this here but I have been looking for a while now. I'm not familiar with Raspberry Pi terminology: what is the difference between split and non-split?