pi1541/Pi1541 #111: Hability tu use CBM font in OLED screen

gbouille, 2018-12-16:
I've change the code to add the hability tu use petscii to display information in LCD via an option.
It works for browsing mode only for the moment.

As i'm a newbe with git, how can I send the code ?

gbouille, 2018-12-17:
Hi all,
Now the CBM font can be use for LCD texts.

Just add an option ( i2cLcdUseCBMChar = 1 ) and the lcd usr the original CBM font if available ( with the option ChargenFont ).
Just be aware that the text is quite small for old eyes C64 early adopter could have now :).

Still don't now how to submit the code. Do I have to do a pull request ? (and how to do this correctly ?)

penfold42, 2018-12-17:
You could double the font size when used on the OLED.

Do you have a branch pushed to your GitHub fork ?

From there it’s a pull request

closed 2018-12-23 by gbouille