pi1541/Pi1541 #112: '?device not present error' when a .d64 image is mounted

kittenpoop, 2018-12-17:
I Created a PI1541 from the following page: https://cbm-pi1541.firebaseapp.com (option A) 

However it is not working correctly, when listing a directory in the "sd2iec mode" it all works perfectly, in such that I'm able to load FB128/FB64 without issues. So I assume that means the connections are correct. (Note: it seems that one sd-card can list the directory ~twice as fast as the other one)

However when I mount a .d64 image it goes bad, producing the error as in the subject. Also when mounting with fb64/128 is hangs on loading forever (the image does get mounted on the PI1541).

I've done the following to see if I can remedy:
1) tried another Raspberry PI (both a 3b+ as a 3b)
2) tried another level converter (now using a robodyn, also tested the sparkfun version)
3) using a raspberry PI power supply (no lightning icon in the upper right corner)
4) Swapped SD card 
5) Used a USB stick to load the d64
6) Also tried option B cable (changing the config.txt)
7) swapped between a C64 and 2 more C128

I do see some activity with the signals in the bottom when it is not working, I see a spike (ATN and CLK). and then no activity. Now using version 01.18, default config.txt (option A).

So I'm out of options here, unless thare is maybe some logging or something to inspect?

(PS: No switches/buzzer/led is connected to the pi)

teknoplay, 2018-12-17:
Have you tried a different power supply? It seems like, when you´re accessing the emulation mode, it starts to work wrong. I built more than 10 "option A" adapters and they work flawlessly. I observed a similar problem with a Raspberry Pi 3b+ and a bad power supply. It started failing as described, and then the PS died.

kittenpoop, 2018-12-17:
Yes, I've used different power supplies and always with the same result. Even bought an original Raspberry Pi supply. And still no go.

kittenpoop, 2018-12-19:
Bought yet another SD card to see if it makes a difference, but it did not..... Now I hooked up a logic analyzer, and hopefully I can see something wrong there. Is there someone that can supply me with the correct (initial) signalling that needs to be done when image is mounted (1541 emulation mode) and a directory is requested for instance.

kittenpoop, 2019-01-01:
Ok, more info... I now have it powerd by a 7.5 Amps meanwell powersupply. No luck. 
Got a pi1541 IO board, still no luck
Looking with a logic analyzer, it seems that the clock is inverted? I see the clock on "HI" at default when signalling (Thus clocking with pulses to low). When I check the youtube video of Steve, it seems inverted with the CLK signal (in SDIEC mode).
And the signals are inverted with the levels indicated on the PI1541's screen. So I think that "true" equals to logic level "0" (LO).
So I experimented with the invert switches in the options.txt, also no luck (using a 7406 btw).

kittenpoop, 2019-01-01:
Wrong button.

closed 2019-01-01 by kittenpoop

sajattack, 2019-03-28:
I'm having the same issue.

sajattack, 2019-03-28:
I think the issue is a bad 1541 rom. I grabbed the one out of vice and it seems to work now.

Swallagoon, 2019-07-19:
Hmm, I'm having the **exact** same issue as described by Kittenpoop. I've tried 3 different ROMs and it's still doing the same thing.

The ROM out of vice didn't work for me. Ugh.

dbills, 2020-01-17:
Sorry I can't offer a solution.  I have this issue also.  Browse mode works, emulation does not.  Mine was a pre-fabbed PCB board.  I've experimented with all suggestions to no avail.  The PCB does not use a level converter ( the components for level conversion are built into the board ) so there is nothing to experiment with there.  One is very tempted to believe the wiring and hardware is correct when the browse works, and that this must be a problem rooted in configuration/rom/software.

proggi64, 2020-07-25:
**Same problem here**: After mounting a d64 file the IEC bus was dead and "device not present".

**Simple solution:**
I used an **8K ROM** image of the original 1541 ROM in the root of the SD card. But that's only half of the code. The pi1541 switches to 6502 emulation when mounting an image and the emulated 6502 crashes because of the missing ROM code. I found a **16K ROM image** for the 1541C and copied it into the root of the SD card (d1541.rom). Now everything works fine.
The SD2IEC mode worked and I thought that the mounting of an d64 should work, too. But that are completely different things. It took me a complete day to understand the problem and solve it.

emartisoft, 2022-06-21:
I cannot mount d64 image file for rpi3 until using 5.1V 2.5A power source.

MatthijsC, 2023-03-06:
Power may be a contributor to the '?device not present error' after a .d64 image is mounted.

I had exactly the same problem. The problem disappeared for me, as soon as:
1. I used a power source of 12W (2.4A iPad power supply)
2. AND: I used a 30cm (11.8 inch) USB cable - I don't have any specs but it's _probably_ rated for a 3A current

luciust, 2025-01-03:
> **Same problem here**: After mounting a d64 file the IEC bus was dead and "device not present".
>
> **Simple solution:** I used an **8K ROM** image of the original 1541 ROM in the root of the SD card. [...]

Confirmed - RPi 3 A runs with lousy CPU, but the problem was the 8kB ROM file - 16kB ROM file from 1514-II