pi1541/Pi1541 #125: Raspberry Pi Model 2 (B) Support

jsalonen, 2019-03-06:
What would it take to add support for Raspberry Pi Model 2 B?

I would be curious to know if thats possible and what are the possible blockers for that.

If its only a matter of contributing, I would gladly try to help out.

jsalonen, 2019-03-06:
Related issue: https://github.com/pi1541/Pi1541/issues/39

aaronnewcomb, 2019-03-27:
I was wondering this as well. I tried it on a 2B with no luck. Won't boot.

pi1541, 2019-11-18:
Please try building for the Pi2 now.
Or download the Pi2 version from my site.
Note: you will probably have to overclock it for better compatibility (currently as of v1.21 the Pi0 (overclocked) is more compatible than a Pi2).

hpingel, 2019-11-25:
I have tested yesterday with an overclocked Pi2 and the latest kernel v1.22 for Pi2. Booting seemed to be unstable for me. From three attempts one attempt works so that I have visible content on the OLED and a complete picture on the HDMI display (I know that HDMI display is not supported on Pi2 but getting a picture there seemed to be reassuring for my first try to play with Pi2).

On the OLED display I got broken character output, when I disabled the usage of the small cbm font, the OLED displayed the lines nicely without corruption.

After mounting a d64 I was not able to read it from the C64. The C64 reacted with "device not present". Maybe further tweaking of the options.txt (GraphIEC=0) will free up more ressources on my Pi2. Will do further testing an report back here. Thank you for supporting this new bunch of Pi devices - this must be horrible to support on the long term. Ideally we need an qemu emulator to emulate Pi1541 on amd64.

hpingel, 2019-11-25:
Currently the type_names in performance.c are filled with "TODO" strings - can I help with filling in values there? Is that easy to update? Not my area of expertise at the moment but I could try.

hpingel, 2019-11-25:
Looking at the PDF comparison table of ThePiHut I see that there are two variants of the Pi 2 Model B. The V1.2 variant is closer to the Pi 3 Model B as it is sharing the ARM Cortex-A53 with it. Just wanted to point that out in case there could be issues with one variant that don't occur with the other.

https://thepihut.com/blogs/raspberry-pi-roundup/raspberry-pi-comparison-table

shazz, 2020-02-14:
Hi, on my side I tried the Pi2 kernel 1.23 on a Rpi 2 Model B V1.1 and at boot I just see the rainbow screen but nothing else.
If copied 
- latest 1541 binaries
- bootcode.bin, fixup.dat and start.elf from the archive link on the website (is it the good firmware ?)
- pi2 1.23 kernel
- dos5141

Anything I could try ?

hobbieman, 2020-05-14:
I installed this on a raspberry pi 2 and made sure to use the kernel image for pi2.  I am able to load most programs, but the ghost-n-goblins fails to load.  I tried it on a pi3 and it loaded just fine, so I know the d64 image of the game is good.

I saw that to get pi1541 working perfectly with pi zero and pi 1, the config.txt needs some modifications.  Are there similar modifications required to get this to work perfectly on pi 2?

pi1541, 2020-05-14:
Try this in your config.txt file in the root of the SD card.

kernel_address=0x1f00000
force_turbo=1
arm_freq=1300
over_voltage=5
gpu_freq=500

# sdram overclock
sdram_freq=500
sdram_schmoo=0x02000020
over_voltage_sdram_p=6
over_voltage_sdram_i=4
over_voltage_sdram_c=4

hpingel, 2020-06-06:
Hi, after a long time I retested one of my Pi 2 Model Bs tonight and it loads and works! I'm using the official kernel downloadable from the website. 

With my Pi2 I can't seem to go higher than arm_freq=1200. Except for that change I am using the parameters provided by Steve.

Some things seem to be disabled in the Pi2 kernel build on purpose to make it run stable? For example the Activity LED doesn't flash like it does with my Pi3.