pi1541/Pi1541 #127: Feature Request: Improve USB keyboard support

MattyBalaam, 2019-04-03:
I have tried a few keyboards which do not get recognised on boot.

1.  An apple keyboard: M2452 (these are from the late 90s)
2. An Apple keyboard: A1243 (these are from about 5/10 years ago)
3. A Numeric keyboard: seems to be rebadged for various people but has the code 'LK-10C' and the brand name 'Eye-T'.

I bought the last one in particular as a nice compact (and quirky looking) solution to cycle through files without having to reach over to the Pi1541, or plug in a big bulky keyboard.

Might be able to find some time to hack-about in the future if there are any pointers about how to be looser in how the software picks keyboards.

penfold42, 2019-04-03:
The USB foundation is provided by:
https://github.com/rsta2/uspi/commits/master

Looks like there was some work done on uspi a few months ago.

n9yty, 2019-06-10:
I also had a failure with a mini USB keyboard (with wireless dongle), an APple keyboard A1243 as you tried, and an old Gateway USB keyboard I had.  So for I have nothing that works no matter which port I use. I saw it mentioned elsewhere that turning on the option for OptionB breaks keyboards, but I haven't tried that and can't confirm it, if I disable it my display doesn't work on the Pi1541.