pi1541/Pi1541 #132: Feature Request - Map a empty button to create a blank/empty disk

kd8bxp, 2019-05-09:
I picked up a hat with a oled, as I'm limited on space - I liked the idea of using the buttons as needed, and being able to see what I'm doing on the OLED.  But creating a new disc from the buttons seems to be missing, so Can the ALT-N (create blank disk) feature be mapped to an empty button (?) 
And if it is already mapped can someone tell me how to use it.

Thanks

bsdf, 2019-05-16:
tangentially related, but would it be possible to create a custom disk/sd2iec command to allow the creation of new d64 images? it would be nice to be able to create new ones using only the c64.

pi1541, 2019-09-22:
> 
> 
> tangentially related, but would it be possible to create a custom disk/sd2iec command to allow the creation of new d64 images? it would be nice to be able to create new ones using only the c64.

This is already implemented. Use the N command.

pi1541, 2019-09-22:
There are no more free buttons to implement this.
Combination clicks are already used for other purposes.
There are plans for supporting extra buttons with IO expanders but this will require extra hardware.

gbelmont22, 2019-09-22:
This is a vague suggestion, but I've been day-dreaming and would like to throw an idea out there and see what folks think.

_Some background:_ I've got an original 1541 case - and my long term goal is to mount my LCD display, rotary controller, some buttons, and a spring-loaded front-panel SD adapter/port on a 5 1/4" face-place, with all the hardware mounted internally in the original drive case.  I mention this because I intend to use the final device as a stand-alone unit with no HDMI/composite video or attached keyboard (just like a 'real' 1541).   

Given those constraints, I'm thinking it might be interesting to develop a 'menu' system that the Pi1541 boots into initially.  So when I flip on the device (or the device gets a reset) this menu is displayed with options like **New Disk**...  **Copy Disk**...  **Mount Disk**...  **Change ROM**...  **Change Device Id**... etc.  The operator would twirl the rotary knob (or click the up/down buttons on the GPIO) and select whatever they need to do.  The existing 'browse for image' logic would probably need a 'back to menu' appended to the bottom of each directory to exit back to the main menu...  Anyway, the code would be most likely be dead simple (I think), because all the heavy lifting is already implemented. - So the feature is really just a 'front-end' that vectors to existing logic.

I'm probably going to do this for my own needs at some point.  But I mention it because it would be a viable way of addressing the need to map a bazillion buttons ... which seems highly impractical.  - Or it might be a _horrible_ idea.  Either way, (if this is an appropriate thread) I'd love any and all feedback (pro or con).   Cheers!