pi1541/Pi1541 #133: 1.20 breaks hardware navigation

bsdf, 2019-05-11:
I tried out the latest version and found that it had completely broken the ability to browse/navigate with my hardware buttons. To illustrate, given the following directory:
```
DISK1.D64
DISK2.D64
DISK3.D64
```
I select `DISK2.D64` with the "down" button and mount it with the "select" button. Once mounted, the display reads:
```
       D1/1
1 DISK2.D64
0>DISK2.D64
DISK3.D64

```
Pushing the "select" button will immediately mount the same disk image again instead of unmounting it. This leaves me in an infinite loop of being unable to unmount the image.

I found that by simultaneously holding "select" and pushing the "down" button allows me to select the next disk image in the directory to mount when releasing "select". The "up" button does not do anything.

Holding "select" and pushing the "directory up" button umounts and goes up a directory and then selects the initial directory again once the "select" button is released.

These issues are not present when reverting back to 1.19

mickstone, 2019-05-11:
I’m seeing the same issue.

wowpin, 2019-05-14:
I can replicate that too. I have noticed that switching the C64 off unmounts the image, although I cannot tell whether this was also a feature in 1.19 and earlier.

bsdf, 2019-05-22:
latest update fixed the issue for me