pi1541/Pi1541 #134: Upper limit on queued disks

rhyddereh, 2019-05-20:
Behavior observed:

Queue (either using lst file or by selecting them) up 9 d64s (Pool of Radiance, 8 sides + save disk).  They seem to load fine but cannot switch between images using the GPIO buttons.

Tested with keyboard and monitor and same behavior, cannot switch - in addition keyboard becomes unresponsive requiring a hard reboot of Pi.

Looks like there is an upper limit that I can’t find documented anywhere.  (Overflowing buffer?)

rhyddereh, 2019-05-20:
I have tested a few disks and it works fine so my setup seems to be working

RWoof, 2019-11-13:
I have this issue as well when trying to play Champions of Krynn, same issue, loading 8 disks plus a save disk from autoswap.lst cause the GPIO up/down buttons to not function anymore. I am able to use the enter/select button to back out of the autoswap.lst and then select the necessary disk on it's own to load that way.

pi1541, 2019-11-17:
Fixed in v1.21

closed 2021-01-11 by pi1541