pi1541/Pi1541 #140: Demo "Artillery 100%" by Shape does not seem to run

we87r6we8, 2019-08-14:
[Artillery 100%](https://csdb.dk/release/?id=93836)

music of demo is playing, but no demo visible...

just a line of information about floppy rom (73, cbm dos v2.6 1541,00,00)

can somebody confirm this?

marksmanuk, 2019-09-29:
Same here (v1.20) although sometimes it runs fine.  Possibly a timing issue?

When the demo runs it does a read from Track 18 and then either hangs playing the background music, hangs with a flashing red border or seems to play fine from that point onwards. 

BTW what a great C64 demo!

grzybsonssg, 2021-03-15:
@marksmanuk @we87r6we8 
I noticed that Artillery 100% works better when using d1541II rom. At least in my case - rpi 3a+, option b hat, firmware v1.20

marksmanuk, 2021-03-16:
I tried the 1541-II ROM.  I didn't know you can switch between drive roms specified in the options.txt file using the buttons, very useful!  It's still very timing sensitive and tends to hang straight after the System and SIDCHIP message, although it seemed to work more often.

The ReadMe file does say it's better on a 1581 or CMD drive.  Music timing to gfx was perfect with the Pi.

Pi1541 v1.24.  I'm using an Action Replay.

Again, that demo is amazing!!!