pi1541/Pi1541 #141: Parallelport for using DolphinDos

pcl6499, 2019-08-27:
Is it possible to make Pi1541, with parallelport / userport option?

penfold42, 2019-08-28:
I don’t think there’s enough GPIO pins on a Pi

andreacampanella, 2019-08-30:
what about an i2c port expander?

penfold42, 2019-08-31:
It’s too slow for the parallel interface.

But, I’d love to see a new hardware variant that moves the buttons and leds to a i2c expander.

That should free up enough for tape emulation or a parallel port.

andreacampanella, 2019-09-01:
or an SPI port expander...

hartkopp, 2020-04-09:
I'm currently running this Pi1541 board https://github.com/hackup/Pi1541io rev4 with a RasPi 3b in "splitted IEC lines" mode and with the OLED display.

Checking either the schematics of the Pi1541io https://github.com/hackup/Pi1541io/blob/master/Pi1541io.pdf and the source code from this repository https://github.com/pi1541/Pi1541/blob/master/src/iec_bus.h#L89 there should be enough pins for a DolphinDos/SpeedDos cable.

The first two columns are the "1541 VIA 6522 hardware pins" form the VIA at $1800 which also handles the IEC bus pins. We could probably assign these pins for the parport cable:

```
-VIA-- -----RasPi GPIO------ pi1541 assignment
02 PA0 08 GPIO14 [TXDO]
03 PA1 10 GPIO15 [RXDO]
## ### 11 GPIO17 [GEN0]      (CLK_BD)
## ### 12 GPIO18 [GEN1]      (DATA_BD)
04 PA2 19 GPIO10 [SPI0_MOSI]
05 PA3 21 GPIO9  [SPI0_MISO]
06 PA4 23 GPIO11 [SPI0_SCK]
07 PA5 24 GPIO8  [SPI0_CE_N]
08 PA6 26 GPIO7  [SPI1_CE_N]
09 PA7 27 GPIO1  [ID_SD]
18 CB1 28 GPIO0  [ID_SC]
39 CA2 31 GPIO6  []
## ### 32 GPIO12 [PWM0]
## ### 35 GPIO19 [SPI1_MISO] (SRQ_BD)

6522 @ $1800
PAx (IN/OUT) data <-> C64 Userport (pin C-L) PBx    (IN/OUT)
CA2 (OUTPUT)       -> C64 Userport (pin B)   /FLAG2 (INPUT)
CB1 (INPUT)       <-  C64 Userport (pin 8)   /PC2   (OUTPUT)
```
So together with the known 3.3v/5v level shifter hardware, we should be able to put an additional parport cable between the pi1541 and the C64 userport ...

I have no know-how in C++ and I'm not really sure whether the current pi1541 Port A emulation of the 6522 VIA at $1800 can handle the data direction switching (@pi1541 ?) which is needed to send/receive the 8-bit data to/from the C64.