pi1541/Pi1541 #151: Can't get Maverick 5.04 to run

bmccord, 2019-11-13:
If I try to use Maverick 5.04 (found here: http://www.haddewig.de/nogames64/tools.html), it will not work.  If I copy the d64 to a real disk and run it on a real drive, it works fine.

On the pi1541, it acts like it's going to load, but then the C64 goes back to basic and says Ready, but everything is pretty well locked up.

I have seen others use Maverick without a problem on the Pi1541.  Could someone else try it and let me know how it works for them?

hpingel, 2019-11-19:
Your problem description is a bit unclear to me. What do you want to achieve?

I have used Maverick (from CSDB I think - not sure now about version number, but I guess 5.04) successfully with Pi1541 and a 1541II.

If you want to write to the Pi1541 from a real disk in a 1541 you need to create a fresh new disk image on the Pi1541 and mount it so that Pi1541 is in emulation mode. Do you have the 7406 on your Pi1541 (option B)?

hpingel, 2019-11-24:
I just retested with Maverick 5.04. I can reproduce the issue you have, but only when I unmount the disk image "Maverick A.d64" too early. You have to leave the disk image mounted and then select "Disk copy". Only when you see the screen where you can change the source/dest device numbers, then you may unmount / eject the Maverick disk.

bmccord, 2019-11-24:
I'm not even getting that far.  I'm not getting to the Maverick menu at all.  Here is a better bug report:

## Steps to Reproduce
1. Mount MBOOTA.D64
2. Type `LOAD "*", 8, 1` and hit `RETURN`
3. The screen goes black for about 2 seconds and then it goes back to a regular basic screen with the word `READY`.
4. Typing anything here is useless.  `RUN` doesn't work.  `LOAD` will not do anything.  `LIST` shows garbage.  You have to turn off the machine and turn it back on.

If I use another copy program to copy the Maverick image (mounted on the Pi1541) to a real disk and then do the above process with the real disk and a real disk drive, I get into Maverick as I would expect (get to the menu).

I am not sure if I have a real 7406.  I bought the kit from ebay and assembled it myself.  The inverter is soldered down and I hate to remove it just to see if the other side has an markings.  I've attached a picture in case you can tell just by looking at the side that is visible.

[image]

Thanks for your help!

hpingel, 2019-11-25:
- What Raspberry Pi are you using? .... Zero 1.2, Zero 1.3, Zero W, 2 B, 2B v1.2, 3B+, 3A+
- Which kernel version of Pi1541 are you using?
- What PSU is used for the Pi?
- What is the Commodore host? C64 (assy?), C128, Plus/4, etc.
- Does the Commodore host have a non-standard ROM? Jiffy, etc. ...
- Which other devices are dangling from your C64 while you run into an issue with your Pi1541?
- What drive ROM file is active on Pi1541?
- Have you tried testing your Pi1541 with bitnax.d64 and/or GraphIEC=0 like suggested here: https://www.lemon64.com/forum/viewtopic.php?p=882574#882574

bmccord, 2019-11-25:
Nice.  I'll look at all those things when I can and get back with answers!  Thanks!

bmccord, 2019-11-26:
Ok.  Here are the answers:

- Raspberry Pi 3B v1.2
- Tried with both kernel 1.20 and 1.22
- The PSU is from CanaKit - 5v 2.5 Amps output
- I have tried both a C64 (breadbin stock, all original) and a C128 (also stock, all original) as far as I know
- Standard ROM - no Jiffy
- I have tried with no other devices attached
- The drive ROM I have used is dos1541 taken from Vice
- I turned GraphIEC to 0 and that didn't help.  So, I tried bitnax.d64 as recommended.  Not sure what it is supposed to look like, but it first comes up with @ABCDEFGHIJKLMNOPQRSTUVWXWZ[£ with a white background.  Then the @ symbol changes to what looks like a red backward 'E'...it's very hard to see what the symbol is.  There doesn't appear to be any drive activity after that.

I also tried:
- The other port on the Pi1541, removing the Pi1541 from HDMI and keyboard, and all of the above on both the C64 and the C128.

Thanks for trying to help!

hpingel, 2019-11-26:
Raspberry Pi 3B v1.2 <-- I guess that is the version without the "plus"?

On forum64.de there is currently a discussion of a couple of Pi1541 users who experience stability problems (Ghosts'n'Goblins Arcade + Star Wars Demo) after upgrading from V1.19/V1.20 to V1.21/V1.22. There are at least two users who just reported that they were able to circumvent stability problems with the latest release(s) by overclocking their Raspberry Pi 3 B's (without the plus) to 1300 MHz.

Source: https://www.forum64.de/index.php?thread/94636-probleme-schwierigkeiten-l%C3%B6sungen-pi1541/&postID=1446958#post1446958

bmccord, 2019-11-26:
Ok.  I'll try to do that tonight and let you know how it goes.

Thanks!

bmccord, 2019-11-27:
Ok.  By overclocking, I got Maverick to load...sometimes.  The bitnax.d64 still doesn't work.  Looks like I may need to get a B+...is that what you would recommend?

Thanks!

hpingel, 2019-11-27:
Don't buy new hardware - try kernel version 1.19 (or older) if that works for you better. As multiple people currently have issues I would recommend to wait until somebody finds the cause and provides a fix. Please follow the thread on lemon64: https://www.lemon64.com/forum/viewtopic.php?t=67978

bmccord, 2019-12-03:
That seems to work much better.  Thanks for your help!

closed 2019-12-03 by bmccord