pi1541/Pi1541 #152: searching for $

elementfan94, 2019-11-18:
I am having this issue with the searching for $. It gets stuck at this and I cant stop it unless I turn off the c64. I know in your instructions to use LOAD"*",8 instead of LOAD"$",8 but it would come up with the device not present error and I would switch to this and it would be stuck on this the whole time. I am usinf the Option B and I have changed the texts in the Options text file to make it work with this. Is this possibly a Rom file error, Options error, or even something wrong with my C64? I have even tried other rom files. Is there one that would be recommended.

I am using a Raspberry Pi 3B, 4GB MicroSd, It is Version 1.21.
Here is my Options text file to see if I'm missing. 
options.txt](https://github.com/pi1541/Pi1541/files/3856314/options.txt)

elementfan94, 2019-11-18:
Update: Redid everything again this time it looks like its reading the LOAD"*",8 but still stuck at Searching for *. Did a test where I unplugged the Pi and on the Commodore It stopped and I got syntax error. So I feel that there something I'm not doing right with the setup or Something not right with the commodore.

pi1541, 2019-11-18:
With option B you need to also add this to your options.txt;-
invertIECOutputs = 1

elementfan94, 2019-11-19:
Still no luck. I got a dead test cartridge to see if possibly something internal inside is the issue. But I have manage to somehow to get it to play on a demo at 1.13 somehow too. So I feel like its possible that I'm still doing something wrong witht the options text.  [options.txt](https://github.com/pi1541/Pi1541/files/3861722/options.txt)

elementfan94, 2019-11-19:
When stuck at the searching for *, I am looking at the Pi through HDMI and the ATN, DAT and ClK are not moving. Assuming this is some oscillator meaning something is not right.

Squirrel61, 2019-11-19:
I don't know if it's related, but I had similar issues on my C64 with an SD2IEC some time ago. A real drive (1541, 1541-II) was working perfectly well, the Pi1541 was working well, a 1541 Ultimate-II was working well, only the SD2IEC was not responding with similar effects like you're having with the Pi1541 (although I didn't analyze the ATN, DAT and CLK lines).

In my case, it turned out that the CIA chip driving the bus was the problem. On my board, the CIA chips where socketed and I simply swapped them, which solved the problems. The funny thing is that there are no other issues after the swapping, so it might have been a timing thing.

So if you're out of other options, check if your CIA chips are socketed and if that is the case, swap them. It might help, it might not, but at least you've tried.

Squirrel61, 2019-11-19:
Btw. I knew the SD2IEC was not at fault because it worked fine on my C128 and even on another C64 which currently has a faulty PLA and RAM chips, it was working.

elementfan94, 2019-11-19:
I'll check that and see if that helps. I also got a dead cartridge test to see if anything needs replacing too.

pi1541, 2019-11-19:
Remove line 11 of the options.txt file in your link, that is;-
invertIECInputs = 1

pi1541, 2019-11-19:
You only need;-
splitIECLines = 1
invertIECOutputs = 1

elementfan94, 2019-11-20:
I made the changes to the options text. Still no luck. I'm just going to wait on a dead test cartridge to see if I have a internal issue on my c64. Squirrel I have opened up my board and the CIA chips are still soldered on possibly the original chips. But I noticed that a lot of the chips are cheap Taiwan to Off name chips. This board has been worked on and whats really interesting is that there are some capacitors double stacking on the board too. Ill see what the cartridge will say but I got a feeling that I'm going to be looking for better chips to use.

elementfan94, 2019-11-21:
Ok, did some more playing and I got it up and running. YAY! But I'm having trouble with the FB I got the FB.d64 on there but when it runs all I see is FB64. I dont see my Programs. I thought I had it in a different directory but its in my 1541 folder I did read your FB64 thing where some original sofware wont work. Is this what's happening?