pi1541/Pi1541 #157: CBM file Browser not working

elementfan94, 2019-11-25:
I am having an issue where I cant read any of the programs on fb.d64. I think this is not how it suppose to work or am I doing something wrong. When reading the directions. It said to access the cbm file browser by using LOAD?*?,8 which I thought was weird. That doesnt work at all. I have a couple programs on the 1541 folder but when I go in all I see is the FB64 and the // in there and not my programs. What am I doing wrong?

penfold42, 2019-11-25:
That’s clearly wrong - looks like some Unicode character screwup on the web page.
“Once the Pi has booted you can simply type LOAD�*�,8 and it will load fb64 (CBM-Browser) and you can use this to navigate folders and select images.”

Try
LOAD “*”,8

elementfan94, 2019-11-25:
Tried that. I get the file not found error.

elementfan94, 2019-11-25:
Heres my Options text to see if Im missing something. Using Option B too. 

[options.txt](https://github.com/pi1541/Pi1541/files/3884794/options.txt)

RWoof, 2019-11-27:
The section I've pasted below from your options file needs to be filled in with the program you want to load by default. Remove the two forward slashes in front of //Starfile and then replace the word somefile with the name of the file you want to load using your load command.

// If you would like to specify what file will be loaded by LOAD"*" in browse mode then specify it here
//StarFileName = somefile

mzeo, 2021-07-20:
I'm having this problem as well. Using option A. Is there any way to try and figure out what is wrong?