pi1541/Pi1541 #158: Raspberry Pi1B GPIO layout

raijin, 2019-11-26:
Hi,

I'm trying to create a pi1541 build on raspberry pi1b model, but pi1b only have 26 gpio. I cant connect my lcd and all buttons, is it possible to have an option to remap the pinout for pi1b?

pi1541, 2019-12-01:
Yes I am going to do this eventually.

Andreas-B1105, 2019-12-10:
@steve: Hi, i bought a pi1541 board (B version with 7406, model by pi1541io) fully assembled with a display and it works pretty good. Now i am trying also to  get to run with an RPI1(26 gpio). I figured that the gpio pins 14 and 15 could be used as pins for RESET (pin38, gpio20) and CLKIN (pin37, gpio26). Those pins are not connected in the hardware, but the miniUART is enabled in your software. Of course i have only a small knowledge yet about the whole software project, but could the the UART simply be recycled to work as the needed I/O pins? There is of course the Button SW5(pin29, gpio05) still missing, but one step after another.
thx Andreas

Andreas-B1105, 2019-12-10:
I forgot to mention, that i built a kernel for RASPPI1 but with ZEROEXPERIMENT =0 and RPIPLUS = 0.
With that the HDMI and the display work, but not the keyboard, not the Buttons and not the emulation of course. I am aware that those function are the minor part, but its only a hint

pi1541, 2019-12-10:
Hi

In that mode the buttons should work.

Yes you should be able to use the UART pins. I keep those pins free as I
use the UART for debug builds so I can trace out stuff and debug it.

Andreas-B1105, 2019-12-12:
Hi rajin, Hi Steve, i don't want to nag you at all. I simply want to inform you, that i 've got one step further. I started a new attempt and changed as follows:
in Makefile.rules i defined 'RASPI' alone. That means the the old version with 26GPIO pins.
To be sure i commented out 'HAS_40Pins'.

in iec_bus.h enum PIGPIO i commented out:
GPIO 26(pin37) PIGPIO_IN_CLOCK
GPIO 20(pin38) PIGPIO_IN_RESET
GPIO 16(pin36) PIGPIO_OUT_LED

and moved so to say those GPIOs to:

GPIO 10(pin19) PIGPIO_IN_CLOCK
GPIO 09(pin21) PIGPIO_IN_RESET
GPIO 11(pin23) PIGPIO_OUT_LED
! On your hardware you have to make those changes also of course !
additionally i changed in rpi-gpio.h
#define LED_GPIO_BIT to 11 (from 16)
then 'make clean', 'make' and copied the new kernel on to my sd card.
The pi1541 boots up(as drive 9), the display shows the directory. 'LOAD"$",9 : LIST let you see this directory on the C64 monitor. Buttons 1-4 work (Button 5 has not been moved yet from GPIO 5(pin27)).
Now here's the 'but'.   
emulation mode does not work, i.e. when you mount an image with button 1 the display shows the emulation mode but, all access attempts with the C64 fail. ('drive not ready' or something similiar on the C64 monitor).
Ok that's it. 
@Steve: If you like, you could tell me how you debug (by using the UART on the RaspPi.

pi1541, 2019-12-13:
You will need to overclock it to work in emulation mode.
To debug add -DDEBUG to the CFLAGS in Makefile.rules (clean and rebuild all) then hook up a serial to USB adaptor to pin 14 TX (it only transmits do you don't need RX). Config your serial software to 115200 baud and no start or stop bits (I think).

Andreas-B1105, 2019-12-31:
Hi All, Success:-)
Overclocking the Rasppi to 900MHz did it. The buttons work and the drive LED.
@ I wish you a good new year and a lot of rain very soon
Thx Andreas

pi1541, 2020-01-01:
Good work!
It may need to put a heat sink on the CPU as overclocking it might make it run hotter.

Correction with serial debugging. You do need one stop bit.

Could you please submit your code changes as a pull request?
Thanks.

Andreas-B1105, 2020-01-03:
OK pull request is created.

idolpx, 2020-04-10:
@Andreas-B1105 is your board an rPi 1BRev2?  I have an rPi 1BRev1 that I'm trying to get working with the Option A wiring but GPIO 2/3 are labeled GPIO 0/1 on this board.  Any tips on getting this working?

idolpx, 2020-04-10:
Well.. I answered my own question.  I changed the GPIO 2 to GPIO 0 and CPIO 3 to GPIO 1 and it's working now.  About to do some more testing but when done I'll post my changes.  :)

Andreas-B1105, 2020-04-11:
Hi, idolpx, i guess i habe the Revision 2. The Gpio pins are not labled at all, therefore I used the Pin Layout from some internet sources. Of course when using the old revision and the GPIO pins 0 & 1 are connected to the physical pins 3 & 5, the software has to be changed too. Good that you found out and you'll post it. It will further clarify the different GPIO Layouts. Thx
Andreas
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit 1&1 Mail gesendet.

idolpx, 2020-04-12:
Yep yep... this reference was useful.

https://elinux.org/RPi_Low-level_peripherals