pi1541/Pi1541 #160: epyx fastloader and TFCIII+ not working anymore

MichaelTroelsen, 2019-12-01:
fastload do not work any more since 1.20. 
I works with kernal 119.

penfold42, 2019-12-01:
Hardware variant ?
Pi model ? Overclock settings ?
PAL/NTSC 64 ?

MichaelTroelsen, 2019-12-01:
PI 3 Model B no overclock.
Pi1541 (option B)
PAL

Might be related https://github.com/pi1541/Pi1541/commit/3978ef21651bae2251efb0bbb19a290cbb48c7b6

pi1541, 2019-12-02:
Please try with 28d1a443c687a1f4e15810d82e9eede718db5a0a

elementfan94, 2019-12-03:
I'm having the same issue too.

MichaelTroelsen, 2019-12-03:
I have build a kernel.img based on current master. The issue has been resolved for TFCIII+ and Epyx Fastloader.

Thraka, 2019-12-26:
I'm using PI 3 Model B no overclocking and Pi1541 (option B) with NTSC and I couldn't get a kernel.img based on master working. Anyone else using NTSC?

rmpel, 2021-04-25:
Reopening/continuing this issue with reason that I have the same problem
Running 1.24 on RPi 3B+
I tried what @MichaelTroelsen did; build from source. The build completes, but the Pi does not work with it. OLED display stays blank, HDMI shows the RasPi color spectrum image. Switching back to 1.24 kernel returns the Pi1541 to working state.

So my main issues is that fast loaders corrupt binary loads. (Directories can be listed, programs cannot be loaded)
I tries many software turbos as well as the Final Cart 3+.

Secondary problem is not being able to build my own kernel.
On Windows, the installation is simple, but the software just doesn't work. Running make in the pi1541-master directory complaints about missing files, which are definitely installed.
On Mac, instructions are not given, but eventually with `brew tap osx-cross/arm` and `brew install arm-gcc-bin` the missing parts are installed and `make` runs without errors (2 warnings, though)

I can also confirm that kernel version 1.19 does NOT show signs of corrupted loads, and fast loading using the FC3+ desktop works.

Any and all assistance to get the latest version working with fast load is very much appreciated.

Thank you!

And thank you for this awesome project. Gave me something electronics to solder, something 3D to remix and print. Love it.

for reference; options.txt
`deviceID = 8
splitIECLines = 1
ROM1581 = 1581-rom.318045-02.bin
scrollHighlightRate = 0.07
ChargenFont = chargen
AutoMountImage = fb.d64
LowercaseBrowseModeFilenames = 1
GraphIEC = 1
SoundOnGPIO = 1
SoundOnGPIODuration = 100
SoundOnGPIOFreq = 200
StarFileName = fb64
LCDName = ssd1306_128x64
LcdLogoName = 1541classic
i2cLcdAddress = 60
i2cScan = 1
QuickBoot = 0
ShowOptions = 0
IgnoreReset = 1`

and config.txt
`kernel_address=0x1f00000 
arm_freq=1100 
over_voltage=8 
sdram_freq=500 
sdram_over_voltage=2 
force_turbo=1 
boot_delay=1`

RetroNynjah, 2021-04-26:
Hi,

I don't have any problems using those fastloaders with 1.24 or a build from current source.
Turn off GraphIEC in options.txt (=0) and see it that helps. I keep it turned off as it ruins compaitibility.

perhof, 2021-04-26:
I don't have any problems using those fastloaders with 1.24 or a build from current source.
Turn off GraphIEC in options.txt (=0) and see it that helps. I keep it turned off as it ruins compaitibility.

rmpel, 2021-04-26:
I will try with the GraphIEC option off tonight, will let you know.

rmpel, 2021-04-26:
followup question, though; "Turn off GraphIEC .... I keep it turned off as it ruins compaitibility."
How does graphing data on screen influence compatibility [with fastloaders]? Should that not be considered a bug, a severe one?

perhof, 2021-04-26:
> followup question, though; "Turn off GraphIEC .... I keep it turned off as it ruins compaitibility."
> How does graphing data on screen influence compatibility [with fastloaders]? Should that not be considered a bug, a severe one?

That statement is based purely on my own experiences and it's not an official statement of any kind. I'm just another user.
I don't know if it's a bug or not but I have noticed that some programs and fastloaders don't work properly with GraphIEC turned on which it is by default in the sample options file. It seems to affect the Maverick Fast Data Copier and some Action Replay features for example.
I figured it's a debug feature that shouldn't be enabled unless you need it. Maybe it's messing with the timings or something.

I don't know if that's even what's causing your problem. It was just a suggestion about something you could try.

rmpel, 2021-04-26:
" It was just a suggestion about something you could try. " 
I know :) And I will certainly try it.

Will let you know and thank you!

rmpel, 2021-04-26:
Turning off the graph seems to work.
Also, incidentally, I managed to build a new kernel using one of my other RasPis, that kernel also requires the graph to be off.
I tested with FC3+ (DLOAD) and with Epyx, fast loading works now! 

Thank you :)

commit 28d1a443 (2019-12-02): Put the bus read and writes back to v1.19 timings.

Fixes the need to overclock a Pi3 for Option A hardware.
Hopefully fixes  #160