pi1541/Pi1541 #167: Is it Possible to Directly Sniff or Capture Raw IEC Bus Traffic?

gitjeff2, 2020-01-04:
Is it possible to directly sniff the IEC bus traffic live (from another device?) or dump it to a file locally on the Pi's SD card? I could see a few possible use cases for this:

1. Capturing an ESC/P or plotter data-stream headed to devices 4&ndash;7. This is my primary interest for asking about this feature.
2. Reverse-engineering of some fast loaders through IEC traffic analysis. I'm sure some modern demo coders would find this handy.
3. Advanced debugging of hardware projects that use the IEC bus.

For live data extraction of bus traffic wouldn't it be possible to echo the stream over USB by optionally treating one of the USB ports as RS232? This would eliminate the need for the receiver to be on a real-time kernel and wouldn't require the use of any additional GPIO pins on the Pi itself.

penfold42, 2020-01-04:
Not for traffic to/from another drive

Just sitting on the bus will let you dump the state of CLK DATA and ATN but you can’t tell who is driving the bus.

You would need to capture the 3 lines on the bus and also capture the 3 lines before the bus drivers.

If the Pi1541 is the only other device on the bus it is possible in theory if it’s fast enough.

gitjeff2, 2020-01-04:
So, basically, I'd have to set the Pi to be device 4 in order to intercept dot-matrix print jobs? I couldn't be another device in the IEC chain passively listening in a manner akin to Wireshark? If I do that, is there a debug-level "capture this traffic" option?

penfold42, 2020-01-05:
Correct

I don’t think gorack has a sniffer in the code but is been a while since I looked at it.

I wrote one that in browse mode (not emulation) would dump their IEC bytes to screen colour coded by send / receive.

You’re better off attaching a logic analyser to the 7406 in the C64. You need 6 wires and probably no more than 10MHz samples

pi1541, 2020-01-11:
You should be able to do this easily no matter what device ID the Pi is and capture the bus activity at 1Mhz.
You will need to buffer all the data in RAM and at some point, dump it to a file or over the UART.
Just hook into the browse mode loop where m_IEC_Commands.SimulateIECUpdate is being called. This loop is being updated at 1Mhz. When capturing the data, stop m_IEC_Commands.SimulateIECUpdate from being called and call your function.
In browse mode m_IEC_Commands.SimulateIECUpdate responds to the ATN and checks the deice ID. Without it the Pi can just passively watch the bus.