pi1541/Pi1541 #169: Cannot Flash ROM to EasyFlash 3

usenethound, 2020-01-09:
Using Kernel 1.23

When directories to locate my cartridge ROMS, Easyflash 3 was not loading directories unless I hit F5 to refresh the listing each time a changed levels.

Once I got to my ROM to flash it to the EasyFlash 3 I pressed enter to flash the ROM (Action Replay 5 NTSC) but there is no response from the keyboard.

I then shut everything down and hooked up my SD2IEC+ and everything worked as it should. All directories loaded and the ROM flashed without issue.

pi1541, 2020-01-11:
You need to put the ROMs into a D64 and mount that.
Pi1541 does not simulate fast loaders when it is browse mode only when it is full emulation mode.

usenethound, 2020-01-11:
Thanks Steve, I will try that this weekend and report back :)

usenethound, 2020-01-18:
Yep the files flash perfectly when put inside a .d64 image and put in emulation mode.

Thanks Steve!

closed 2020-01-18 by usenethound