pi1541/Pi1541 #170: .g64 Multidisk support BUG?

usenethound, 2020-01-09:
Kernel 1.23

When adding multidisk g64 images either using the pihat buttons or a keyboard hooked up to the pi and adding them with the insert key I'm getting graphical glitches and crashes. It happens when you enter the shops and start adding mods and once you hit the shop on the far right sometimes it crashes or just gives graphcal glitches when you are racing after you click to load side 2.

I'm trying the following images from the c64 Preservation Project
4x4_offroad_racing_s1[epyx_1988](ntsc)(!).g64
4x4_offroad_racing_s2[epyx_1988](ntsc)(!).g64

This happens with or without a Fastloader cartridge inserted

If I just load 4x4_offroad_racing_s1[epyx_1988](ntsc)(!).g64 by itself without adding the second image to the set, I get no crashes or graphical bugs when entering the 3 different shops.

I haven't tried creating a .lst file yet

pi1541, 2020-01-11:
NTSC or PAL?

usenethound, 2020-01-11:
NTSC

usenethound, 2020-01-11:
Updated my original post with the proper filenames for the images. Github formatting messed up the original filenames

usenethound, 2020-01-17:
Any luck reproducing this Steve?

mutetus, 2020-01-19:
I got this issue after 1.19, where everything just randomly started glitching and demos freezing when loading a .lst file and trying to load anything. I tried to locate the problem and successfully fixed the issue in 1.23 by adding nullptr checks to DiskImage object in DiskCaddy::ShowSelectedImage and DiskCaddy::Update. Don't know the actual source of the problem, I'm just too busy watching demos using .lst files right now.

usenethound, 2020-01-19:
mutetus, first and foremost, thanks for sharing this information. Hopefully it will help Steve track down the issue!

would you mind sharing your build?

mutetus, 2020-01-19:
[Pi1541-1.23.zip](https://github.com/pi1541/Pi1541/files/4083053/Pi1541-1.23.zip)
Here's a copy of my build folder, though the only altered source file is DiscCaddy.cpp. There are still some weird issues with lists, such as entries with too long names/spaces or something not shown when list is loaded and they show up when the files are renamed. I didn't investigate this any deeper, I must have spent less than an hour getting familiar with the code base before I tried the fix and there's no way I could debug this stuff, but I hope this gives you some kind of clue what goes wrong with the lists. Cheers and excellent work so far!

usenethound, 2020-01-19:
mutetus, thanks for that. Hopefully Steve can figure this out and include the fix in his next build.

Fingers crossed......

usenethound, 2020-01-19:
I can confirm that this fixes the corruption I was getting. I'm not sure what other side effects the patch by mutetus may have yet but everything was working fine with the 4x4 ntsc .g64 from the preservation project.

Thanks again mutetus

I'll leave this open so Steve can see where the issue lies and fix it in his next build.

pi1541, 2020-01-25:
I could not reproduce the problem.
I have committed a possible fix.

usenethound, 2020-01-25:
Steve, I built a new kernel.img from your latest source and the issue actually got worse. The entire screen was corrupted after loading side 2. I rolled back to 1.19 and everything works perfect again. I'm at a loss but willing to help in any way I can if you care to pursue this issue further. I know it's darn near impossible because you can't reproduce the issue on your end.

pi1541, 2020-01-25:
Does mutetus' DiscCaddy.cpp work for you?
Please email me at pi1541contact@gmail.com

commit 4cf352d7 (2020-01-26): Fixed #170
closed 2020-01-26 by pi1541