pi1541/Pi1541 #171: PI 3A+  Not working

john-clark, 2020-01-19:
Same SD card working in Pi 3b+ but 3A+ it is not working.

Krnel version of Pi1541 testing [Latest Version 1.23 (Pi3)] https://cbm-pi1541.firebaseapp.com/kernel.zip

What happens is it gets stuck on the rainbow color bios boot screen.

Is there hardcoding for USB hub or ethernet that causes the kernel to fail? Do I need a different kernel?

pi1541, 2020-01-21:
It works for me.
And I just tried with the latest firmware (start.elf, bootcode.bin and fixup.dat) from the pi foundation.

john-clark, 2020-01-22:
I did the same. I took the card from the 3B+ where it works fine and then put the same card into the 3A+ and it does not work. Here is a video demonstrating the problem https://youtu.be/zR4wEiadwlo

I am using PI files from firmware-1.20180919,zip  WinVICE-3.2-x86.7z  d1541II and the above kernel.zip

edolnx, 2020-02-01:
I'm seeing the same problem. My Pi 3A shows the PCB revision of V1.1 - I've used a Pi 3A before but this one just gives the rainbow screen. I don't know what revision the board was that worked, I gave it to a friend.

pi1541, 2020-02-01:
Does anyone have serial to USB that can help debug this?

edolnx, 2020-02-02:
I have a USB Serial adapter, and I'm not getting anything of the RX and TX pins on the GPIO header when trying to boot the 3A. But I also get nothing off the working 3B board, so maybe my adapter is bad or I'm missing something...

pi1541, 2020-02-02:
You need to build a debug build (-DDEBUG) and you only need TX.
Please email me at pi1541contact@gmail.com.

edolnx, 2020-02-15:
Tried building with `-DDEBUG` instead of `-NDEBUG` in the `Makefile.rules` and it failed with the following errors. Any ideas?

```
  LINK kernel
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: uspi/libuspi.a(uspilibrary.o): in function `USPiInitialize':
uspilibrary.c:(.text.USPiInitialize+0x30): undefined reference to `uspi_assertion_failed'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: uspilibrary.c:(.text.USPiInitialize+0x250): undefined reference to `uspi_assertion_failed'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: uspi/libuspi.a(uspilibrary.o): in function `USPiKeyboardAvailable':
uspilibrary.c:(.text.USPiKeyboardAvailable+0x3c): undefined reference to `uspi_assertion_failed'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: uspi/libuspi.a(uspilibrary.o): in function `USPiKeyboardRegisterKeyPressedHandler':
uspilibrary.c:(.text.USPiKeyboardRegisterKeyPressedHandler+0x48): undefined reference to `uspi_assertion_failed'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: uspilibrary.c:(.text.USPiKeyboardRegisterKeyPressedHandler+0x74): undefined reference to `uspi_assertion_failed'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: uspi/libuspi.a(uspilibrary.o):uspilibrary.c:(.text.USPiKeyboardRegisterShutdownHandler+0x48): more undefined references to `uspi_assertion_failed' follow
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: uspi/libuspi.a(usbconfigparser.o): in function `USBConfigurationParserError':
usbconfigparser.c:(.text.USBConfigurationParserError+0x44): undefined reference to `DebugHexdump'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: usbconfigparser.c:(.text.USBConfigurationParserError+0x5c): undefined reference to `uspi_assertion_failed'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: usbconfigparser.c:(.text.USBConfigurationParserError+0x78): undefined reference to `uspi_assertion_failed'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: uspi/libuspi.a(usbdevicefactory.o): in function `GetDevice':
usbdevicefactory.c:(.text.GetDevice+0x14c): undefined reference to `uspi_assertion_failed'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: usbdevicefactory.c:(.text.GetDevice+0x168): undefined reference to `uspi_assertion_failed'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: usbdevicefactory.c:(.text.GetDevice+0x188): undefined reference to `uspi_assertion_failed'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: uspi/libuspi.a(usbdevicefactory.o):usbdevicefactory.c:(.text.GetDevice+0x214): more undefined references to `uspi_assertion_failed' follow
collect2: error: ld returned 1 exit status
make: *** [Makefile:22: kernel] Error 1
```

pi1541, 2020-02-15:
You still need -DNDEBUG use this;-
CFLAGS	 += $(ARCH) -Wall -Wno-unused-variable -Wno-unused-but-set-variable -Wno-psabi -fsigned-char -fno-builtin -Ofast -DNDEBUG  -DDEBUG

edolnx, 2020-02-16:
It builds with those parameters, but I still don't get any output on the serial port (GPIO PIN8) with the kernel.img built on a Pi3B+ or the Pi3A+

edolnx, 2020-02-25:
So, I discovered the root of my problem. What I thought was a 3A+ was just a A. Swapped it out with a 3A+ and everything is running great. Sorry about that.

john-clark, 2020-02-27:
Updated ticket to indicated my issue is same as edolnx. Wondering if we can make another image for the 1a+.  Thanks!

john-clark, 2020-03-04:
I bought a pi zero wireless, pi3a+ and pi4. Could not get pizero wireless working with the pi0 image, did not try the pi4. I would really like this working on a pi1a+ due to form factor.

Closing this issue since 3a+ works fine.

closed 2020-03-04 by john-clark