pi1541/Pi1541 #172: Buzzer sound gets stuck with an annoying 1kHz steady signal

derryx, 2020-01-28:
Hi!

I am using V1.23 on an RPi3B+. Everything works fine but when I enable sound with the following settings

`SoundOnGPIO = 1`
`SoundOnGPIODuration = 100 // Length of buzz in micro seconds`
`SoundOnGPIOFreq = 200 // Frequency of buzz in Hz`

I get head movement sounds but after some time the sound gets stuck with an annoying 1kHz sound. Pi1541 still works fine when this happens but the sound does not go away until I reboot.

CU
Thomas

maccasoft, 2020-02-13:
I had the same problem. It turned out it was a wrong piezo speaker, one that has an internal oscillator that emits sound when powered, while this project requires a simple speaker. The firmware is only partially responsible of the issue because it happens that the sound gpio pin is left at the logic 1 state powering the speaker.
After replacing the speaker with the correct type all worked fine.
The speaker was part of a kit purchased on e-bay so I guess others may have received the wrong type.

I'm attaching a patch that fixes the issue, or those who can't replace the speaker.

[gpio_sound_stop_patch.txt](https://github.com/pi1541/Pi1541/files/4198774/gpio_sound_stop_patch.txt)

derryx, 2020-02-14:
Thank you very much. That was really the problem! I purchased my kit on eBay as well. I contact the seller.

closed 2020-02-14 by derryx

dlxmax, 2024-07-09:
@pi1541 It'd be great for everyone if this code was merged!