pi1541/Pi1541 #175: Can't get Golden Axe (NTSC) to load properly.

RWoof, 2020-02-22:
I've tried a couple versions, one from Nostalgia (https://csdb.dk/release/?id=33518), it loads to a fastloader screen, then eventually pauses on a black screen forever or dumps me back to the command prompt. Another release I tried was a nbz file from the C64 preservation project, this one loads quickly to some copy protection that seems to work, when it tries to load the game I end up locked on a black screen again. One time with this version I made it to a garbled screen but no further. I tried to copy the game file to a real floppy using dracopy but the disk would not list or load properly.

pi1541, 2020-03-07:
The Nostalgia version you linked works for me. Please try "GraphIEC = 0" in your options.txt file.

RWoof, 2020-03-07:
I tried, I can see their crack screen, then it goes to a warp loader, then the trainer screen, when I press space to get past the trainer (without setting any options to yes) it loaded until track 22 and then stopped on a black screen. The second time I tried, same thing except it went to track 23 and dumped me back to the ready prompt. Third time I tried I got to track 23 and then stuck on the black screen again. Fourth time I tried I got stuck at a black screen again on track 21. Maybe my VIC II is incompatible somehow? I have a 6567R9. I've attached my options file with the change you suggested.

[options.txt](https://github.com/pi1541/Pi1541/files/4301407/options.txt)

RWoof, 2020-03-07:
I tried again and got to a frozen title screen, that's the farthest I've been so far.

RWoof, 2021-02-01:
Downloaded 1.24 from the pi1541 website, failed to load anything at all (other than fb64), load errors on every game I tried. Compiled kernel.img myself in linux after cloning the github and it works fine. It was a bit smaller than the file on the website. Also plays Golden Axe now so perhaps one of the commits fixed it for me! Thanks!

closed 2021-02-01 by RWoof