pi1541/Pi1541 #185: Rotary encoder - swap quadrature pins

penfold42, 2020-06-24:
From lemon64:

popeye2755 wrote:
Is it possible to reverse the way the rotary encoder works? Possibly via the options.txt file?

penfold42 wrote:
Nope - not that I can see looking through the source. It would be an easy change to implement. In the meantime can you not just swap the 2 wires ? 

popeye2755 wrote: 
The encoder is mounted on the PCB. It seems that there are two different types, one with a green base and one with a blue base. They work in opposite directions.

gbelmont22, 2020-08-13:
Is this simply a matter of swapping the CLK/DATA pin mapping or would it be better to swap the up/down logic coming out of the decoder?  

Or, asked another way:

The rotary encoder has an 'init' method that tells the code what GPIO pins are connected to the CLK/DATA pins of the encoder.  If the physical CLK/DATA pins are reversed on a 'green' vs. 'blue' encoder, then it would be dead simple for me to add a 'invert' option and simply flip the init logic.  By swapping the CLK/DATA pins it should simply reverse the operation of the encoder.  But I'd like some confirmation of my interpretation of the OPs problem before I implement anything...

penfold42, 2020-08-15:
The pin swapping is the way to go - the PR looks good