pi1541/Pi1541 #196: Action replay 6 Fastload or Epyx cannot find FB64 on pi1541

lovrenco, 2020-10-19:
Although I can load FB64 from Jiffydos (I use it as rom on EF3), if I try to load with Epyx Fastload or Action replay 6 Fastload they hang on "SERCHING FOR FB64".

Can anyone test and confirm this?

I planed to use AR6 for save states in games, but problem I just described + no possibility to format disk from AR6 is really a problem.

CaIcipher, 2020-10-20:
I have probably the same issue. C64 assy 250407, Epyx Fastload reloaded. raspberry pi 3b. 3A power supply, pi1541 v1.23, 1541 
Tested also different ROMs.

When I hit Commodore+Run/Stop --> freezes at "Searching for 0:*"
Load"$",8 works but when I try to load FB64 again -->  freezes at "searching for FB64"

Regardles which PRG file I try to load, does not work as long as the Epyx Fastload Module is attached. When I unplug the module I can load PRG files from the pi1541

If I mount a disk image using the pi1541 buttons then loading FB64 or any other PRG file works fine.

perhof, 2020-11-14:
> Although I can load FB64 from Jiffydos (I use it as rom on EF3), if I try to load with Epyx Fastload or Action replay 6 Fastload they hang on "SERCHING FOR FB64".
>
> Can anyone test and confirm this? [...]

I haven't seen the problem you are describing while using AR6 (genuine cartridge). FB64 works fine for me.
Are you using the FB64 version that has been modified for Pi1541? That's the one from the pi1541 project homepage [https://cbm-pi1541.firebaseapp.com/fb.d64](https://cbm-pi1541.firebaseapp.com/fb.d64)

I have seen some issues (hanging) when using the AR6 fastloader while having JiffyDOS loaded on the Pi1541. On Ghost and Goblins Arcade for example.
If I switch to stock 1541 ROM on the Pi1541 the same things work without problems. Sure, there's no reason to run Jiffy on the Pi1541 when using the AR6 but I'm still curious as to why they don't work together. Jiffy is said to behave just like the stock ROM unless you also run a Jiffy (or Jaffy) kernal on the computer but there seems to be a difference.
Do you have the same problems loading FB64 if you switch to a stock 1541 ROM on the pi1541?

nfonseca, 2020-12-29:
Hi !

I am getting the same problem with a c64 Assy 250466 and a Epyx Fast Load.
I cannot get CBM loaded while the cartdrige is inserted.

Thanks

Pitbull6571, 2021-04-27:
I had the same problem with my C64. The file browser doesn't like the fast loader unless you place a disk image named fb.d64 into the 1541 folder. The fb.d64 disk image has to contain the fb64 program file. Then you must set you options.txt file to auto load that disk image upon startup and then make fb64 the file to open with LOAD "*",8 .... Honestly the only thing I use the file browser for is dracopy when I copy games and files to real floppy's. I have an HDMI monitor and USB keyboard for browsing the files. Puts less wear on my commodore and so far all fast loaders I have tried work without issue. SD2IEC devices need a file browser. Any of the pi1541's worth having don't need a file browser!