pi1541/Pi1541 #201: Pi1541 cannot finalize D64 disk copy written by Maverick 5.04

fcprindle, 2020-12-01:
As documented on Nov 24, 2020 in this forum post:
[https://www.lemon64.com/forum/viewtopic.php?t=76426&start=18](url)
If Pi1541 is used to flip between the source and backup diskettes while running Maverick 5.04, option A (Single 1541) **AND** the Pi1541 ROM1 is either of the standard CBM 1541 ROMs (dos1541 or d1541II), then Maverick will experience write errors on the backup diskette and the resulting backup diskette (after finalization via Pi1541 button 1) will have zero length.

**But** if the Pi1541 ROM1 happens to be Jiffy.bin (JiffyDos), the Maverick copy will complete without write errors and the resulting backup diskette (after finalization via Pi1541 button 1) will be a normal 174,848 byte D64 file. The copy will be perfect.

I haven't been able find the bug that causes this to happen, but did determine that the file ends up truncated to zero length because the finalization code in DriveImage.cpp member function WriteD64() will be unable to find sector 0 of track 18 anywhere in the tracks array (populated in emulation mode). This problem is reproducible using the Maverick 5.04 disk image supplied in this forum post:
[https://www.lemon64.com/forum/viewtopic.php?t=76426&start=16](url)

Note that under exactly the same scenario and using one of the standard CBM 1541 ROMs and true drive emulation, _VICE_ will **NOT** experience write errors with the Maverick 5.04 single 1541 copy and will produce a proper 174,848 byte D64 file for the backup diskette. I did notice that while writing to the backup diskette, the Tape indicator at the bottom of the GTK3VICE-3.4-win64-r37296 GUI flickers; this may or may not be a significant observation.

perhof, 2021-01-02:
I have tested the referenced commit with Maverick 5.04 dual 1541 and it does seem to fix these problems.
I was hoping that this fix would solve similar problems with other versions of Maverick too but unfortunately it doesn't. I still experience problems with this version for example: [https://csdb.dk/release/?id=144938](https://csdb.dk/release/?id=144938)

pi1541, 2021-01-02:
This is like 5.04 single drive copy.
I can't even get it to work in Vice.
[image]

fcprindle, 2021-01-03:
Honest to God, I just tried this with VICE using this version of Maverick from Tom-Cat (I can't tell what version it is): http://pc.sux.org/tomcat/pack.zip
I tried it with JiffyDOS ROMs and I tried it with stock kernal/dos1541 ROMS, and in both cases the single drive copy worked perfectly. This is the same version of Maverick that fails single drive copy with Pi1541 if its ROM1 = dos1541.

perhof, 2021-01-03:
I hadn't tried it before but I just tried the Black/Angels 1999 version in Vice now and it works just fine for me.
Copying both with single drive and dual drive from device 8 to device 9 without errors, Tried emulating devices both as 1541 and 1541-II drives but I can't reproduce the problem in Vice.

[image]

closed 2021-01-06 by pi1541

perhof, 2021-01-06:
> I have tested the referenced commit with Maverick 5.04 dual 1541 and it does seem to fix these problems.
> I was hoping that this fix would solve similar problems with other versions of Maverick too [...]

I got some more time to try these newer builds today and the problem for me was that I unknowingly had GraphIEC enabled in my config. Sorry about that.
Commit 1c6093d fixes my problem too if GraphIEC = 0.
Thanks!

fcprindle, 2021-01-06:
I'll pull the latest source and give Maverick another try.

perhof, 2021-01-06:
> I got some more time to try these newer builds today and the problem for me was that I unknowingly had GraphIEC enabled in my config. Sorry about that.
> Commit [0ae6208](https://github.com/pi1541/Pi1541/commit/0ae620842f293bbbbb4a6301082d2aa1f1ee8da0) fixes my problem too if GraphIEC = 0.
> Thanks!

Another positive effect of the latest change is that the disk utilities in Action Replay 6 are working now.
Copying and formatting of disks is working just fine. Never got those things working before.

fcprindle, 2021-01-06:
Yes, I've just tested and the original issue that Tom-Cat discovered is resolved with this change!

crystalct, 2022-02-08:
@pi1541 Hi Steve, i don't know your mail so i try this way to ask you a thing.
Theoretically speaking, can a Rapsberry to emulate a MOS 8726 and then a REU Cartridge?

Thanks and compliments for your great work here!!!

commit 0ae62084 (2021-01-06): Fixed #201 with regards to Maverick's single drive copy.