pi1541/Pi1541 #205: Emulation mode navigation in FB broken in 1.24/Pi0/Epyx FL

teknoplay, 2021-01-23:
When using Epyx Fast Load, with new firmware 1.24, trying to load FB64 will hang the C64.
Hardware is RPI0.
Not happening in 1.23.
My options.txt file is set to automount fb.d64 just to force Emulation mode and avoid compatibility problems with FL.
The activity led is not working either.
both things work in 1.23 with same options file.

gibbsjj, 2021-02-01:
Same issue here on an RPi 3B + original Epyx FL cart + v1.24.

imbezol, 2021-11-16:
Same issue here with RPi 3B + Super Snapshot V5.22 cartridge + v1.24. Works fine on 1.23.

gestorum, 2021-12-31:
Same issue on RPi zero 2 + Epyx Fastload cartridge + v1.24.

Jynx73, 2022-04-10:
Same issue here with RPi 3B+, 1.24 and Epyx FL (1984) and loading from the boot screen. Works if I disable FL before trying to load.