pi1541/Pi1541 #206: Activity light doesn't work on Pi Zero with v1.24 software

tyristori, 2021-01-25:
Works fine on v1.23.

Jartza, 2021-01-25:
I can confirm this issue, updated kernel.img.to 1.24 and activity led stays off, went back to 1.23 and activity led works again.

fcprindle, 2021-01-28:
When I run on a Pi Zero with a kernel built from the latest master branch, the activity LED generally seems to work fine. I haven't actually downloaded the pre-built 1.24 kernel.img (nor the 1.24 source branch) since I wanted whatever was the latest.

However, in one instance, after loading the game _Raid on Bungeling Bay_, the activity LED remained on, and in fact remained on after I reset the C64 and loaded the directory successfully, and even after I reset the drive by powering off/on the C64. I needed to power down the pi1541 and then power it back up to get the LED to turn off.

perhof, 2021-04-14:
I'm seeing an issue that might be related.
The activity LED doesn't seem to work in 1.24 on Pi3 either when SplitIECLines=0
Going back to 1.23 and the activity LED works fine again using the same configuration.

It works fine in 1.24 too as long as SplitIECLines=1

fcprindle, 2021-04-14:
Yes, my Pi Zero hat is option B, so I **do** have SplitIECLines=1 and as I said, am seeing the activity LED working. So this 1.24 bug would seem to be independent of the Pi model, but dependent on the option.

vari77, 2021-05-02:
My activity led doesn't blink with 1.24, instead with 1.23 the led blinks correctly.
I'm using a Pi Zero with hat option A, so SplitIECLines=1 is commented.
In my case it was enough to replace the "kernel.img" for solve the problem.

kotbehemot53, 2021-11-06:
Any updates on this? I can confirm that in my case with 1.24, Pi Zero, SplitIECLines=0, it doesn't work either.

perhof, 2021-11-06:
I got some test builds emailed to me from Stephen in June but none of them made any difference.
He didn't have an option A hat with activity LED to test with and I offered him one of my builds but I got the impression he didn't want it and then he disappeared. No more answers to my emails.

Warshi7819, 2022-06-22:
So, after I built the simple Pi1541 build (option A) I ran into this problem as well in the latest version. So what do you do? Well, it turns out that this project has some excellent information on how to setup the build environment on Windows (my prefferd platform for some reason we shall not dig too deep into) and with some knowledge from the internet like "it worked in version 1.23" it was fairly easy to find the breaking change, revert it and compile a new "1.24" kernal while we wait on the true Gurus to release v.1.25 and the offical fix.

https://github.com/pi1541/Pi1541/blob/master/src/iec_bus.h
lines: 304 - 316. Revert the change meaning the section should look like this afterwards:

		if (!splitIECLines)
		{
			// This means that when any pin is turn to output it will output a 0 and pull lines low (ie an activation state on the IEC bus)
			// Note: on the IEC bus you never output a 1 you simply tri state and it will be pulled up to a 1 (ie inactive state on the IEC bus) if no one else is pulling it low.

			myOutsGPFSEL0 = read32(ARM_GPIO_GPFSEL0);
			myOutsGPFSEL1 = read32(ARM_GPIO_GPFSEL1);

			myOutsGPFSEL1 |= (1 << ((PIGPIO_OUT_LED - 10) * 3));
			myOutsGPFSEL1 |= (1 << ((PIGPIO_OUT_SOUND - 10) * 3));
			//RPI_SetGpioPinFunction((rpi_gpio_pin_t)PIGPIO_OUT_SOUND, FS_OUTPUT);
			//RPI_SetGpioPinFunction((rpi_gpio_pin_t)PIGPIO_OUT_LED, FS_OUTPUT);
		}

You can also see me ramble about how I fixed/patched it here:
https://youtu.be/InrRrrOgxqw

And how to setup the development environment here:
https://youtu.be/tXXUUnhUf1M

morphinejh, 2024-01-21:
@Warshi7819  - your 'solution' solves both the light and the buzzer issues on the Raspberry PI Zero built with Option A (SplitIECLines=0). Thanks!

Haven't experienced any ill effects yet.

vari77, 2024-01-21:
Dear morphinejh, could you share the compiled version for Pi0? Thank you!

morphinejh, 2024-01-21:
@vari77 This is a, **use at your own risk** situation as it is not fully tested. However, I have not had any issues as of yet.

The only changes to the the source are the two commented lines and two un-commented lines listed above in the `iec.h` file. Make sure to rename the image file.
[pi1541_zero_1.24Hack_forBuzzerandLED_use-at-your-own-risk.zip](https://github.com/pi1541/Pi1541/files/14002125/pi1541_zero_1.24Hack_forBuzzerandLED_use-at-your-own-risk.zip)

vari77, 2024-01-21:
Hi @morphinejh, I'll try it on my Pi0, thanks a lot!

lalcaraz, 2024-02-15:
@morphinejh I tried your IMG and works fine ... except that SD2IEC compatibility is broken. I.E. I can't play SIDs using the sd2iec version. It works on regular v1.24. But at least the LED and Piezo are back.