pi1541/Pi1541 #214: Software that uses FLOAD2.1 crashes when using the fload load routine. pi3/1.24

Retroisten, 2021-05-26:
Software that uses FLOAD2.1 as loader crashes when using the fload load routine on a pi3/1.24

With 1.23 everything is fine.

As an example two releases which use this loader:

https://csdb.dk/release/download.php?id=246925
https://csdb.dk/release/download.php?id=246577

perhof, 2021-06-04:
I can't reproduce this in 1.24 on any of my Pi3's.
Would you mind sharing your options file?

Retroisten, 2021-06-04:
Sure. 

[options.txt](https://github.com/pi1541/Pi1541/files/6601004/options.txt)

perhof, 2021-06-04:
Nope. Still can't seem to reproduce it.
I downloded your options.txt and used it as is. Mounted both the above d64 imges and loaded the games. Both games starting fine. I could even cycle through all tracks of 8bit-Slicks.
Are you using any other tweaks or specific methods to produce the error?
I guesst that fast loader could be quite timing sensitive. Is Ghosts'n Goblins Arcade (https://csdb.dk/release/?id=139257) loading properly all the way or is it producing a garbled title screen?

Retroisten, 2021-06-05:
It has a garbled screen with 1.24. With 1.23 the title screen is ok. No special mods. Strange.

perhof, 2021-06-05:
Sounds like a timing issue then. Maybe the hex inverter (7406?) is bad or the wrong type?
There are some warnings about using parts from unreliable sorces on the homepage if that could be the case?
https://cbm-pi1541.firebaseapp.com/
It's also possible that turning off GraphIEC in options would help here. It didn't cause the problem you are having when I tried it on my hardware so I still think your 7406 could be marginal.

LazeMSS, 2022-12-27:
Same problem - using kernel 1.23 - no problem - using 1.24 cause garbage on Ghost and Goblins