pi1541/Pi1541 #220: Piezo Buzzer - certain loaders force the buzzer to stick 'on' with high pitched squeal

billlagr, 2021-06-25:
Hi,
A few specific loaders (eg, the Castlevania .g64) cause the buzzer to stick on with an ear splitting high pitched squeal. Only way to stop the noise is to power cycle the Pi, and then disable the buzzer either through a jumper or in the config file before loading.
Is it possible to add a way to shut the buzzer off dynamically through the keyboard or a button combination - eg ALT-S to toggle the buzzer?
This issue has been present for a while, I'm currently using 1.24 but I know it has happened in earlier versions too. 99% of images load fine and the buzzer works correctly, it's just a particular few.

glbyers, 2021-12-12:
I'm experiencing the same issue, and have tested with a Pi2B and a Pi3A with numerous hats. I tested versions 1.24 back to version 1.21 and they all experience it. When it occurs, i'm seeing a little over 3V on the GPIO pin 33. There are no shorts between pin 33 and any other pin, so this seems to be Pi1541 not pulling down when stepper is off. Can be triggered repeatedly with the "Uncensored" demo from Booze design, for example.

glbyers, 2021-12-12:
And... This only occurs when using a Jiffy DOS ROM.

ismal, 2022-10-05:
It depends on the buzzer! I bought two different ones and I has this problem (looks like they were "active" once) and then I switched to a small "passive" one from China and it works great.